libimobiledevice/libimobiledevice #806: Check restore is done

Hiepars, 2019-06-25:
How to check if restore is done?
thanks

nikias, 2019-06-25:
Depends on what exactly you want to detect, and how. idevicerestore at some point will receive a success message knowing the restore is finished, while from the outside, e.g. with ideviceinfo, you could maybe tell if the version number changed. Won't help if a restore was performed to the same version though. From a libimobiledevice perspective though the device will disappear from normal mode, and reappear in normal mode. If restore failed it will stay in recovery mode.

Hiepars, 2019-06-25:
thanks for your answer.
i am trying to restore a full backup package. 
i get message: 
The device should reboot now.
Restore Successful.

But on Iphone screen, restore is still in progressing.
I want to check restore is completely done.

Hiepars, 2019-06-26:
Restore always in normal mode then i do not have any idea to check restore is done.

Hiepars, 2019-06-26:
[image]
this is iPhone screen when restore is done. how to check iPhone is displaying this screen?

mexmer, 2019-06-26:
afaik this information is not available trough cable.

Blefish, 2019-06-26:
I haven't been able to detect this either. But it might be possible to check if device is booted via springboardservices, for example, by querying the icons.

Dantee296, 2019-06-27:
@Hiepars you can find this info probably in purplebuddy or springboard / backboard 
you gotta need to search for it.

Hiepars, 2019-06-27:
@Dantee296 
1 - purplebuddy
how can i get this file : Com.apple.purplebuddy.plist ?
this file is in backup package so i can not get it if restore is not finished!

sKerry-not-scarry, 2021-03-02:
Hi @Hiepars, were you able to find any solution?

a-gunter, 2021-03-02:
`ideviceinfo -q com.apple.PurpleBuddy` will show information that will only get updated once springboard starts up. You can compare the output during the restore process vs once the "Restore Completed" screen is shown, and most likely there will be a difference somewhere

Hiepars, 2021-03-03:
> Hi @Hiepars, were you able to find any solution?

> `ideviceinfo -q com.apple.PurpleBuddy` will show information that will only get updated once springboard starts up. You can compare the output during the restore process vs once the "Restore Completed" screen is shown, and most likely there will be a difference somewhere

closed 2021-03-03 by Hiepars