libimobiledevice/libimobiledevice #807: iOS13 problems with SSL_read and select. Proposing solution.

JonGabilondoAngulo, 2019-06-26:
Hi,
I have written a tool using house_arrest and AFC to copy files into an App. In iOS3 it stopped working. Analysing the problem I found that the SSL_read in "idevice_connection_receive_timeout" may be not correct for SSL reads. The read loop does a select first and then SSL_read after. Now that services are SSL enabled, and is some cases the encryption does take place, like in house_arrest, we can see the problem more frequently.
It is documented that the implementation of select/read and select/SSL_read are different:
https://www.openssl.org/docs/man1.0.2/man3/SSL_read.html
https://stackoverflow.com/questions/10294018/integrating-ssl-into-a-program-which-uses-bsd-sockets

In the current implementation a "afc_receive_data" would get blocked and time out will be triggered. With this change, following the recommended patter for select/SSL_read, my tool is successful in iOS13. (The select should be performed after SSL_rad and only if SSL requests is, otherwise may carry blocking situations.)

I'll be glad if this is considered correct that a change to "device.c" "idevice_connection_receive_timeout" would be considered in the master branch.

```
LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_receive_timeout(idevice_connection_t connection, char *data, uint32_t len, uint32_t *recv_bytes, unsigned int timeout)
{
	if (!connection || (connection->ssl_data && !connection->ssl_data->session) || len == 0) {
		return IDEVICE_E_INVALID_ARG;
	}
    
	if (connection->ssl_data) {
		uint32_t received = 0;

		while (received < len) {

/* OLD WAY
			int conn_error = socket_check_fd((int)(long)connection->data, FDM_READ, timeout);
			idevice_error_t error = socket_recv_to_idevice_error(conn_error, len, received);

			switch (error) {
				case IDEVICE_E_SUCCESS:
                    debug_info("IDEVICE_E_SUCCESS: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
					break;
				case IDEVICE_E_UNKNOWN_ERROR:
					debug_info("ERROR: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
				default:
                    debug_info("ERROR: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
					return error;
			}
*/

#ifdef HAVE_OPENSSL
			int r = SSL_read(connection->ssl_data->session, (void*)((char*)(data+received)), (int)len-received);

// NEW WAY
           int ssl_err = SSL_get_error(connection->ssl_data, r);
            switch (ssl_err) {
                case SSL_ERROR_WANT_READ: {
                    int conn_error = socket_check_fd((int)(long)connection->data, FDM_READ, timeout);
                    idevice_error_t error = socket_recv_to_idevice_error(conn_error, len, received);

                    switch (error) {
                        case IDEVICE_E_SUCCESS:
                            debug_info("IDEVICE_E_SUCCESS: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
                            break;
                        case IDEVICE_E_UNKNOWN_ERROR:
                            debug_info("ERROR: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
                        default:
                            debug_info("ERROR: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
                            return error;
                    }
                }   break;
                case SSL_ERROR_NONE: {
                } break;
                default: {
                   debug_info("ERROR from SSL_read. At:%s Error:%s", "idevice_connection_receive_timeout", ssl_error_to_string(ssl_err));
                } break;
            }
#else
			ssize_t r = gnutls_record_recv(connection->ssl_data->session, (void*)(data+received), (size_t)len-received);
#endif
			if (r > 0) {
				received += r;
			} else {
				break;
			}
		}

		debug_info("SSL_read %d, received %d", len, received);
		if (received < len) {
			*recv_bytes = 0;
			return IDEVICE_E_SSL_ERROR;
		}
		
		*recv_bytes = received;
		return IDEVICE_E_SUCCESS;
	}
	return internal_connection_receive_timeout(connection, data, len, recv_bytes, timeout);
}
```

linpeifei, 2019-07-09:
error: use of undeclared identifier 'FDM_READ'
           int conn_error = socket_check_fd((int)(long)connection->data, FDM_READ, timeout);
error: static declaration of 'ssl_error_to_string' follows non-static declaration
           static const char *ssl_error_to_string(int e)

Could you tell me how to solve the problem above?Please.

nikias, 2019-07-14:
@JonGabilondoAngulo with this approach the first SSL_read() could block, that's what the commit 4d8b89223cbc9f530cc650ab5131c09eab1af258 tried to solve.

JonGabilondoAngulo, 2019-07-17:
Here a simpler solution that works for me while using select and not doing the first SSL_read potentially blocking.
In afc.c when it reads the message of 48 bytes, first SSL_reads the header of 40 bytes, then SSL_reads the remaining 8 bytes. Doing select before the second SSL_read causes select to reach timeout, select doesn't see any bytes in the transmission buffer, they seem to have been emptied by the first SSL_read.

The following fix solves my AFC problem while keeping select. I do "select" optional by checking if SSL_pending has bytes ready to consume.

`LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_receive_timeout(idevice_connection_t connection, char *data, uint32_t len, uint32_t *recv_bytes, unsigned int timeout)
{
	if (!connection || (connection->ssl_data && !connection->ssl_data->session) || len == 0) {
		return IDEVICE_E_INVALID_ARG;
	}
    
	if (connection->ssl_data && !connection->bypass_ssl) {
        uint32_t received = 0;
        bool do_select = true;
        
		while (received < len) {

#ifdef HAVE_OPENSSL
            do_select = SSL_pending(connection->ssl_data->session) == 0; // In SSL the message might have been fully read already, if the SSL buffer has the bytes dont do select.
#endif
            
            if (do_select) {
                int conn_error = socket_check_fd((int)(long)connection->data, FDM_READ, timeout);
                idevice_error_t error = socket_recv_to_idevice_error(conn_error, len, received);
                
                switch (error) {
                    case IDEVICE_E_SUCCESS:
                        debug_info("IDEVICE_E_SUCCESS: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
                        break;
                    case IDEVICE_E_UNKNOWN_ERROR:
                        debug_info("ERROR: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
                    default:
                        debug_info("ERROR: socket_check_fd returned %d (%s)", conn_error, strerror(-conn_error));
                        return error;
                }
            }
`

nikias, 2019-07-18:
See commit f52584e7310ad9af414cdd22cbfad81d53417c22.

JonGabilondoAngulo, 2019-07-22:
Great. Closed.

closed 2019-07-22 by JonGabilondoAngulo

abidkhansn, 2022-03-01:
Hi JonGabilondoAngulo,
Your tool which writes files into iDevices using House Arrest and AFC services, works with iOS 15? I have such a tool which is not working with iOS 15.