libimobiledevice/libimobiledevice #810: your ssl cert has expired

Lision, 2019-06-30:
$ brew install --HEAD usbmuxd
==> Cloning https://git.sukimashita.com/libusbmuxd.git
Updating /Users/lision/Library/Caches/Homebrew/usbmuxd--git
fatal: unable to access 'https://git.sukimashita.com/libusbmuxd.git/': SSL certificate problem: certificate has expired
Error: An exception occurred within a child process:
  DownloadError: Failed to download resource "usbmuxd"
Failure while executing; `git fetch origin` exited with 128. Here's the output:
fatal: unable to access 'https://git.sukimashita.com/libusbmuxd.git/': SSL certificate problem: certificate has expired

Lision, 2019-06-30:
$ brew install --HEAD libimobiledevice
Updating Homebrew...
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/lision/Library/Caches/Homebrew/libimobiledevice--git
fatal: unable to access 'https://git.libimobiledevice.org/libimobiledevice.git/': SSL certificate problem: certificate has expired
Error: An exception occurred within a child process:
  DownloadError: Failed to download resource "libimobiledevice"
Failure while executing; `git fetch origin` exited with 128. Here's the output:
fatal: unable to access 'https://git.libimobiledevice.org/libimobiledevice.git/': SSL certificate problem: certificate has expired

danielporto, 2019-06-30:
Can confirm. Cannot install libimobiledevice in my system too. :
fatal: unable to access 'https://git.libimobiledevice.org/libimobiledevice.git/': SSL certificate problem: certificate has expired

wongpiwat, 2019-06-30:
Me too. This is my error :  Cloning into '/Users/user/Library/Caches/Homebrew/libimobiledevice--git'...
fatal: unable to access 'https://git.libimobiledevice.org/libimobiledevice.git/': SSL certificate problem: certificate has expired

NielsB95, 2019-06-30:
I've got this error as well! Hopefully, you can fix this soon :).

Lision, 2019-07-01:
I just found out that it has been solved, thx!

closed 2019-07-01 by Lision