libimobiledevice/libimobiledevice #813: idevicecrashreport not working on iOS 13

h4x4g0n1337, 2019-07-11:
Using the latest code to date will result in what looks like a hang
Debugging the issue reveals that the code gets stuck while waiting for a "ping" from crashreportmover service.
I was able to fix the issue by adding the following code to idevicecrashreport.c
```
...
    /* trigger move operation on device */
    idevice_connection_t connection = NULL;
    device_error = idevice_connect(device, service->port, &connection);
    if(device_error != IDEVICE_E_SUCCESS) {
        lockdownd_client_free(lockdownd);
        idevice_free(device);
        return -1;
    }

// FIX:
    if (service->ssl_enabled) {
        device_error = idevice_connection_enable_ssl(connection);
        if (device_error != IDEVICE_E_SUCCESS) {
            fprintf("Enabling ssl failed. device_error: %d\n", device_error);
            lockdownd_client_free(lockdownd);
            idevice_free(device);
            return -1;
        }
    }
// END OF FIX

    /* read "ping" message which indicates the crash logs have been moved to a safe harbor */
    char *ping = malloc(4);
    memset(ping, '\0', 4);
    int attempts = 0;
...
```

The core issue is that lockdownd tells us the service will use SSL. The code is able to detect this but does not respect it properly.

Another issue exists while trying to read the crash logs.
I was able to fix the 2nd issue by following @JonGabilondoAngulo instructions in issue #807

nikias, 2019-07-12:
Fixed in a different way with commit 2332655423c1616d8e37ece7f33e98be0e218504.

h4x4g0n1337, 2019-07-13:
wouldn't it be better to fix it in "idevice_connect" by inspecting the "ssl_enabled" property and calling the enable ssl function?

closed 2019-08-17 by h4x4g0n1337