libimobiledevice/libimobiledevice #817: Changing to TLS_method is not universal！

xdeng, 2019-07-24:
in iOS7.1.2 
SSL_CTX_new(TLS_method()); 
SSL_do_handshake Will fail
Lower versions may also fail

I tried TLSv1_method and the compatibility would be better.

xdeng, 2019-07-24:
openssl 1.1.0.5

nikias, 2019-07-24:
Can you get me the correct OpenSSL version? The version must be something like 1.1.0j or 1.1.0k

xdeng, 2019-07-25:
1.1.0e

nikias, 2019-08-10:
TLSv1_method has been deprecated in 1.1.0 so I don't know how this would even compile...

nikias, 2019-08-10:
and TLS_method is defined as
```
#define SSLv23_method           TLS_method
```

xdeng, 2019-08-12:
@nikias   but  iOS7.1.2 I tried to use TLS_method() and the handshake will fail       you can try try

elekven-m, 2019-08-14:
Fails on iOS 8.3 also, same place. Built with mingw32, running on Win10.

nikias, 2019-08-15:
I just tried and it fails indeed. Looking into it...

elekven-m, 2019-08-15:
Replacing "TLS_method" with "TLSv1_method" in idevice.c and ignoring the function deprecation warnings makes it work with all iOS versions both old and 13.0 beta.

elekven-m, 2019-08-15:
Using TLSv1_1_method also works on older iOS versions. However TLSv1_2_method fails. These are also deprecated. If using gcc, can surround usage with #pragma GCC diagnostic ignored "-Wdeprecated-declarations" (with push/pop of same to maintain state) to eliminate the warning message.

Also, in idevice.c, move the #include <windows.h> to below the #include socket.h to eliminate the "must include winsock.h before windows.h" complaint.

elekven-m, 2019-08-16:
Hello Nikias,

Replacing "TLS_method" with "TLSv1_method" in idevice.c and ignoring the function deprecation warnings lets it work with all iOS versions from 7 to 13.0 beta. I don’t have anything older than 7 to test with.

A better long term solution would be to attempt to first establish the SSL_CTX with a newer handshake method, and if it fails then back off to TLSv1_1 in case the iOS is older. That way if the newer higher security algorithm is accepted by a newer iOS things are good, while older iOS can still work with the old security algorithm.

TLSv1_method: works all iOS versions
TLSv1_1_method: works all iOS versions
TLSv1_2_method: fails on iOS < 9
If using gcc, can surround usage with #pragma GCC diagnostic ignored "-Wdeprecated-declarations" (with push/pop of same to maintain state) to eliminate the warning message:
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wdeprecated-declarations"
              SSL_CTX *ssl_ctx = SSL_CTX_new(TLSv1_1_method());
#pragma GCC diagnostic pop
Having the function deprecated in openSSL doesn’t mean it’s missing, it just means new code shouldn’t use it. But we are tied to old iOS devices, so we must use it since they do not know how to do the more advanced negotiations.
Also, in idevice.c, move the #include <windows.h> to below the #include socket.h to eliminate the "must include winsock.h before windows.h" complaint.
Thanks for working on all this libimobiledevice code.
Eric Lekven

nikias, 2019-08-18:
I found a better solution by explicitly disallowing TLS 1.1, 1.2, and 1.3 (if supported by OpenSSL/LibreSSL) so it should only try with TLS 1.0. See commit a9ccd31ddd5d59db8dc5114c7c85e3eac8743578.
@elekven-m I also moved the windows.h header, see commit 9e33a26264ab6abe9eea10b7aa11f948b4bde87e.

mexmer, 2019-08-28:
@nikias  i did instead following

SSL_CTX_set_max_proto_version(Context, SSL3_VERSION) for ios < 9
it does work without problems.

nikias, 2019-08-29:
See my comment here: https://github.com/libimobiledevice/libimobiledevice/issues/833#issuecomment-526084191

mexmer, 2019-08-29:
based on my testing, you need to set both MIN and MAX vetsion to TLS1, it will rule out devices with IOS 3 and bellow, but even ios 13 works with that.

i suspect, there might be bug in openssl, that if you set MAX but not MIN, it will still try to negotiate hightest version, although it can't (seem openssl problem, but i'm not in mood debuging openssl), settings both works. and that's what we need.

mexmer, 2019-08-29:
it's quite interesting that ios 3 doesn't work with TLS 1, since TLS 1.0 was standardized on 1999, while ios came out 10 years later ... but that's apple ... just works (or not) 😸

mexmer, 2019-09-04:
just update on this. i retested TLS1 MIN/MAX with iphone 2g (1st gen iphone) running ios 3.1.2 without jailbreak. phone did successfully handshake.

from that we can conclude TLS1 MIN/MAX in openSSL works across all current iphones (from IOS 3 - which was last for 1st gen iphone), up to IOS 13 (which is newest upcoming iOS version)

i would still suggest to leave max unset for higher versions of ios (like ios 10 or newer)

nikias, 2019-09-04:
Ok can you confirm setting only MIN to TLS1 works with 10+ ?

mexmer, 2019-09-05:
min set to TLS_1 max not set
ios 3.1.2 doesn't work
ios 4.2.1 doesn't work
ios 5.1.1 doesn't work
ios 12.4 works
ios 13.1 works

those that doesn't work fail with SSL_handshake again.

nikias, 2019-09-05:
Ok thanks for confirming. I can change the code accordingly. But now I wonder how older OpenSSL would work in this scenario, since SSL_CTX_set_min_proto_version + SSL_CTX_set_max_proto_version was first introduced in OpenSSL 1.1.0 - in this case, I am using the currently implemented way to restrict the versions; I guess I can adapt it with the same version check and only allow > TLS 1.0 on iOS 10+ - makes sense?

mexmer, 2019-09-05:
nikias tbh. on older openssl, i used TLS1_method for new devices and SSLv23_method for old devices.

with 1.1 you can set MIN to TLS1 and set max TLS1 for iOS < 10, and not set max for ios 10+

mexmer, 2019-09-05:
for SSL 1.0, you can probly use

SSL_CTX_set_options(SSL_OP_NO_TLSv1_1 | SSL_OP_NO_TLSv1_2);

nikias, 2019-09-05:
```
diff --git a/src/idevice.c b/src/idevice.c
index 382e9d2..fd1f4b5 100644
--- a/src/idevice.c
+++ b/src/idevice.c
@@ -320,7 +320,7 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_connect(idevice_t device, uint16_t
 		new_connection->type = CONNECTION_USBMUXD;
 		new_connection->data = (void*)(long)sfd;
 		new_connection->ssl_data = NULL;
-		idevice_get_udid(device, &new_connection->udid);
+		new_connection->device = device;
 		*connection = new_connection;
 		return IDEVICE_E_SUCCESS;
 	} else {
@@ -348,9 +348,6 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_disconnect(idevice_connection_t con
 		debug_info("Unknown connection type %d", connection->type);
 	}
 
-	if (connection->udid)
-		free(connection->udid);
-
 	free(connection);
 	connection = NULL;
 
@@ -759,9 +756,9 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_enable_ssl(idevice_conne
 #endif
 	plist_t pair_record = NULL;
 
-	userpref_read_pair_record(connection->udid, &pair_record);
+	userpref_read_pair_record(connection->device->udid, &pair_record);
 	if (!pair_record) {
-		debug_info("ERROR: Failed enabling SSL. Unable to read pair record for udid %s.", connection->udid);
+		debug_info("ERROR: Failed enabling SSL. Unable to read pair record for udid %s.", connection->device->udid);
 		return ret;
 	}
 
@@ -789,16 +786,27 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_enable_ssl(idevice_conne
 		return ret;
 	}
 
-	/* force use of TLSv1 */
+#if OPENSSL_VERSION_NUMBER < 0x10100002L || \
+	(defined(LIBRESSL_VERSION_NUMBER) && (LIBRESSL_VERSION_NUMBER < 0x2060000fL))
+	/* force use of TLSv1 for older devices */
+	if (connection->device->version < DEVICE_VERSION(10,0,0)) {
 #ifdef SSL_OP_NO_TLSv1_1
-	int opts = SSL_OP_NO_TLSv1_1;
+		long opts = SSL_CTX_get_options(ssl_ctx);
+		opts |= SSL_OP_NO_TLSv1_1;
 #ifdef SSL_OP_NO_TLSv1_2
-	opts |= SSL_OP_NO_TLSv1_2;
+		opts |= SSL_OP_NO_TLSv1_2;
 #endif
 #ifdef SSL_OP_NO_TLSv1_3
-	opts |= SSL_OP_NO_TLSv1_3;
+		opts |= SSL_OP_NO_TLSv1_3;
+#endif
+		SSL_CTX_set_options(ssl_ctx, opts);
 #endif
-	SSL_CTX_set_options(ssl_ctx, SSL_OP_ALL | opts);
+	}
+#else
+	SSL_CTX_set_min_proto_version(ssl_ctx, TLS1_VERSION);
+	if (connection->device->version < DEVICE_VERSION(10,0,0)) {
+		SSL_CTX_set_max_proto_version(ssl_ctx, TLS1_VERSION);
+	}
 #endif
 
 	BIO* membp;
diff --git a/src/idevice.h b/src/idevice.h
index 94e828b..f403c55 100644
--- a/src/idevice.h
+++ b/src/idevice.h
@@ -46,6 +46,8 @@
 #include "common/userpref.h"
 #include "libimobiledevice/libimobiledevice.h"
 
+#define DEVICE_VERSION(maj, min, patch) (((maj & 0xFF) << 16) | ((min & 0xFF) << 8) | (patch & 0xFF))
+
 enum connection_type {
 	CONNECTION_USBMUXD = 1
 };
@@ -66,7 +68,7 @@ struct ssl_data_private {
 typedef struct ssl_data_private *ssl_data_t;
 
 struct idevice_connection_private {
-	char *udid;
+	idevice_t device;
 	enum connection_type type;
 	void *data;
 	ssl_data_t ssl_data;
diff --git a/src/lockdown.c b/src/lockdown.c
index 23f2782..694fb47 100644
--- a/src/lockdown.c
+++ b/src/lockdown.c
@@ -715,7 +715,7 @@ LIBIMOBILEDEVICE_API lockdownd_error_t lockdownd_client_new_with_handshake(idevi
 			char *s_version = NULL;
 			plist_get_string_val(p_version, &s_version);
 			if (s_version && sscanf(s_version, "%d.%d.%d", &vers[0], &vers[1], &vers[2]) >= 2) {
-				device->version = ((vers[0] & 0xFF) << 16) | ((vers[1] & 0xFF) << 8) | (vers[2] & 0xFF);
+				device->version = DEVICE_VERSION(vers[0], vers[1], vers[2]);
 			}
 			free(s_version);
 		}
@@ -738,7 +738,7 @@ LIBIMOBILEDEVICE_API lockdownd_error_t lockdownd_client_new_with_handshake(idevi
 	plist_free(pair_record);
 	pair_record = NULL;
 
-	if (device->version < 0x070000) {
+	if (device->version < DEVICE_VERSION(7,0,0)) {
 		/* for older devices, we need to validate pairing to receive trusted host status */
 		ret = lockdownd_validate_pair(client_loc, NULL);
 
```

mexmer, 2019-09-06:
thanks, i will probly not have time until end of week, so i can confirm this monday or tuesday.

nikias, 2019-09-06:
I think I can safely commit it. If it doesn't work, we can fix it later.

nikias, 2019-09-06:
See commit a9e69b1252e5918b6d8ada1209ccefde301cfa26.