libimobiledevice/libimobiledevice #818: Is there a document of all keys

chiyuelaochao, 2019-07-26:
Hi sir,

Thank for your attention.
I get IMEI uses the following function with the key of "InternationalMobileEquipmentIdentity":
lockdownd_get_value(ilock, NULL, "InternationalMobileEquipmentIdentity", &value);
and gets Serial Number by this function with the key of "SerialNumber":
lockdownd_get_value(ilock, NULL, "SerialNumber", &value);

I only know a few keys, is there a document of all keys?

Best Regards,
Wade

Blefish, 2019-07-26:
All of the keys come from mobilegestalt it seems, but only some of them are available via lockdownd as well. 

'ideviceinfo -k key' vs 'idevicediagnostics mobilegestalt key'

When you ask via mobilegestalt, it can be queried by both the original string and md5 hash value.

These are the keys I've found, but many are available only via mobilegestalt.
https://blog.timac.org/2017/0124-deobfuscating-libmobilegestalt-keys/
https://blog.timac.org/2018/1126-deobfuscated-libmobilegestalt-keys-ios-12/

chiyuelaochao, 2019-07-26:
Thanks a lot.
It really helps me.

Best Regards
Wade

chiyuelaochao, 2019-07-28:
Hi sir,

One more question. Thanks for your time.
I use this function with the key "DeviceEnclosureColor" to get phone enclosure color.
lockdownd_get_value(ilock, NULL, "DeviceEnclosureColor", &value);
It returns the value appears to be an integer.
For example, the value is 1 for a black iPhone 7 and it's 8 for a coral iPhone XR.

How these integer values represent phone enclosure colors?
Or is there any other keys or functions can get phone enclosure colors correctly.

Best Regards,
Wade

Blefish, 2019-07-28:
Check this: https://gist.github.com/Blefish/ecdbd21341f66a0a53095bd84bc2a1d0

There is an unofficial list of the devicecolor,enclosurecolor=color mapping but it is not very up to date. However you can find out the color if you follow the gist, as the images loaded have the right color :)

chiyuelaochao, 2019-07-29:
Thank you very much, it gives me the right tips although it's unofficial documentation.

The following are values represent phone enclosure colors from iPhone xs max to iPhone 7 I find, hope it can help more people.

iPhone XS Max
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,6-1-1-0/online-infobox__3x.png	Space Gray
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,6-1-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,6-1-4-0/online-infobox__3x.png	Gold

iPhone XS
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,2-1-1-0/online-infobox__3x.png	Space Gray
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,2-1-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,2-1-4-0/online-infobox__3x.png	Gold

iPhone XR
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,8-1-1-0/online-infobox__3x.png	Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,8-1-2-0/online-infobox__3x.png	White
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,8-1-6-0/online-infobox__3x.png	Red
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,8-1-7-0/online-infobox__3x.png	Yellow
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,8-1-8-0/online-infobox__3x.png	Coral
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone11,8-1-9-0/online-infobox__3x.png	Blue

iPhone X
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,3-1-1-0/online-infobox__3x.png	Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,3-1-2-0/online-infobox__3x.png	White
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,6-1-1-0/online-infobox__3x.png	Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,6-1-2-0/online-infobox__3x.png	White

iPhone 8 Plus
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,2-1-1-0/online-infobox__3x.png	Space Gray
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,2-1-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,2-1-3-0/online-infobox__3x.png	Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,2-1-6-0/online-infobox__3x.png	Red
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,5-1-1-0/online-infobox__3x.png	Space Gray
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,5-1-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,5-1-3-0/online-infobox__3x.png	Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,5-1-6-0/online-infobox__3x.png	Red

iPhone 8
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,1-1-1-0/online-infobox__3x.png	Space Gray
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,1-1-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,1-1-3-0/online-infobox__3x.png	Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,1-1-6-0/online-infobox__3x.png	Red
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,4-1-1-0/online-infobox__3x.png	Space Gray
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,4-1-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,4-1-3-0/online-infobox__3x.png	Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone10,4-1-6-0/online-infobox__3x.png	Red

iPhone 7 Plus
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,4-1-1-0/online-infobox__3x.png	Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,4-2-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,4-2-3-0/online-infobox__3x.png	Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,4-2-4-0/online-infobox__3x.png	Rose Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,4-1-5-0/online-infobox__3x.png	Jet Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,4-2-6-0/online-infobox__3x.png	RED

https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,2-1-1-0/online-infobox__3x.png	Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,2-2-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,2-2-3-0/online-infobox__3x.png	Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,2-2-4-0/online-infobox__3x.png	Rose Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,2-1-5-0/online-infobox__3x.png	Jet Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,2-2-6-0/online-infobox__3x.png	RED

iPhone 7
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,1-1-1-0/online-infobox__3x.png	Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,1-2-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,1-2-3-0/online-infobox__3x.png	Gold
[... cut ...]
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,3-2-2-0/online-infobox__3x.png	Silver
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,3-2-3-0/online-infobox__3x.png	Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,3-2-4-0/online-infobox__3x.png	Rose Gold
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,3-1-5-0/online-infobox__3x.png	Jet Black
https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,3-2-6-0/online-infobox__3x.png	RED

closed 2019-07-29 by chiyuelaochao

HuangTao15521192185, 2019-10-12:
Does it mean that I can use the command "ideviceinfo -k OLEDDisplay" to get information about OLED?

ganvinalix, 2021-08-13:
好东西

KhalsaSarav, 2023-07-27:
Hi chiyuelaochao,

thanks for sharing this valuable information. 
I need your help to where I get same information regarding iPhone 13, iPad 4 generation or some other iOS devices. 

Actually, I want to know the process where you got  these image URLs 'iPhone9,1-2-6-0'  ("https://statici.icloud.com/fmipmobile/deviceImages-9.0/iPhone/iPhone9,1-2-6-0/online-infobox__3x.png") and where you got color code value. 

I want to fix it for all devices but  some of these are physicals not available. 

Regards & thanks 
Sarav (cellDe)