libimobiledevice/libimobiledevice #820: installation_proxy gets timeout listing Apps. Solved.

JonGabilondoAngulo, 2019-07-28:
Hi,
got reports that listing applications "ideviceinstaller -l" on devices with 100+ Apps was not returning all apps list.  Installing Apps was reported to fail too. I checked and I could reproduce it.

The timeout occurred in "instproxy_receive_status_loop" , calling "property_list_service_receive_plist_with_timeout". Increasing the timeout to 5000 has solved the problem. 

Does it make sense to you ? Can it be a fix to add to master ?

`static instproxy_error_t instproxy_receive_status_loop(instproxy_client_t client, plist_t command, instproxy_status_cb_t status_cb, void *user_data)
{
	instproxy_error_t res = INSTPROXY_E_UNKNOWN_ERROR;
	int complete = 0;
	plist_t node = NULL;
	char* command_name = NULL;
	char* status_name = NULL;
	char* error_name = NULL;
	char* error_description = NULL;
	uint64_t error_code = 0;
#ifndef STRIP_DEBUG_CODE
	int percent_complete = 0;
#endif

	instproxy_command_get_name(command, &command_name);

	do {
		/* receive status response */
		instproxy_lock(client);
		res = instproxy_error(property_list_service_receive_plist_with_timeout(client->parent, &node, 5000)); // 1000 default, 5000 solves problems
		instproxy_unlock(client);
`

JonGabilondoAngulo, 2019-09-22:
In case it helps anyone .. the problem is not the timeout. It is that 'res' can be ENOENT !
The loop should not end if ENOENT. 
It doesn't mean there is an unrecoverable communication problem. Keep the loop going.

res = instproxy_error(property_list_service_receive_plist_with_timeout(client->parent, &node, 1000));

nikias, 2019-09-22:
I don't think it's ENOENT. Is the value -2? Then it's INSTPROXY_E_PLIST_ERROR. Please verify.

JonGabilondoAngulo, 2019-09-23:
I was using a libusbmuxd without the fix for ETIMEDOUT of Jun 13. My fault.

closed 2019-09-23 by JonGabilondoAngulo