libimobiledevice/libimobiledevice #824: can't install

Adolf-L, 2019-08-06:
OSX  ~/Desktop/libimobiledevice-master  ./autogen.sh
glibtoolize: putting auxiliary files in '.'.
glibtoolize: linking file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: linking file 'm4/libtool.m4'
glibtoolize: linking file 'm4/ltoptions.m4'
glibtoolize: linking file 'm4/ltsugar.m4'
glibtoolize: linking file 'm4/ltversion.m4'
glibtoolize: linking file 'm4/lt~obsolete.m4'
./autogen.sh: line 15: aclocal: command not found
./autogen.sh: line 17: automake: command not found
./configure: line 2329: syntax error near unexpected token `dist-bzip2'
./configure: line 2329: `AM_INIT_AUTOMAKE(dist-bzip2 no-dist-gzip check-news)'

Kanong017, 2019-08-13:
[](url)[[](url)]([url]([url]([url]([url]([url]([url]([url]([url]([url]([url]([url]([url]([url]([url]([url]([url]( www.google.com)))))))))))))))))

nikias, 2019-08-13:
you need to install autoconf and automake and other dependencies... this is mentioned in the README.md

closed 2019-08-13 by nikias