libimobiledevice/libimobiledevice #825: iOS 12.1 - ideviceImageMounter fails to mount: returns error code -3 (MOBILE_IMAGE_MOUNTER_E_CONN_FAILED)

MidiGlitch, 2019-08-06:
Using an iPad with iOS 12.1 installed. Am getting an error when using iDeviceImageMounter. The reading of the images looks to have passed ok, but when trying to mount the image, it returns almost immediately with a MOBILE_IMAGE_MOUNTER_E_CONN_FAILED error.

Gorz94, 2020-08-26:
Hello,

Did you find a solution ? I have the same problem.

nikias, 2020-08-26:
Can you run `idevicesyslog -q` on another terminal while trying to mount and post the output somewhere?

Gorz94, 2020-08-26:
Yeah, sure.

[iPad.txt](https://github.com/libimobiledevice/libimobiledevice/files/5128936/iPad.txt)

Thanks for helping

nikias, 2020-08-26:
```
Aug 26 11:36:16 mobile_storage_proxy[221] <Notice>: Failed to receive secure message: [54: Connection reset by peer]
Aug 26 11:36:16 mobile_storage_proxy[221] <Notice>: Failed to receive data.
Aug 26 11:36:16 mobile_storage_proxy[221] <Notice>: Failed to receive bytes into <private>.
Aug 26 11:36:16 mobile_storage_proxy[221] <Notice>: Failed to receive bytes.
Aug 26 11:36:16 mobile_storage_proxy[221] <Notice>: Sending response: <private>
Aug 26 11:36:16 mobile_storage_proxy(liblockdown.dylib)[221] <Notice>: _send_message: <private>
Aug 26 11:36:16 mobile_storage_proxy[221] <Notice>: Failed to send message: 0x00000001
```
Seems like ideviceimagemounter disconnects. Can you run `ideviceimagemounter` with `-d` (make sure you compiled libimobiledevice with `--enable-debug`)?

Gorz94, 2020-08-26:
Actually, I use windows binaries found on the net. I cannot achieve to build this library on my computer, and my build of ideviceimagemounter doesn't has the "--enable-debug". I will try again to compile it.

Gorz94, 2020-08-26:
I just found 2019 binaries of libimobiledevice, and I just mounted correclty my ddi, I really don't understand.

nikias, 2020-08-26:
Well, then the answer is simple: Whoever build them didn't build them correctly.

Gorz94, 2020-08-26:
That's probably the main reason yeah.
However, do you know if there is any Apple legal restrictions about mounting a DDI on a device ? And what does happen if we update the ios version ?

nikias, 2020-08-26:
Mounting is not legally restricted but the DDI is intellectual property of / copyrighted by Apple so you cannot distribute it if that's what you want to know. If you update the iOS version you need the matching DDI for that version.

Gorz94, 2020-08-26:
Yeah, that was what I was asking for, thanks.
If you mount the version N on an iOS device, then you update it to the N+1. Mounting the N+1 DDI version will work correctly ? 
That is my last question, promised ! :D

nikias, 2020-08-26:
Yes, every version has its own DDI and yes it will work.

Gorz94, 2020-08-26:
Ok. Thanks for answering !

lightcode4, 2020-08-27:
This might be unrelated but I have am issue would like to share with everyone here 

So first I have an iPhone x and while trying to jailbreak it keeps returning error.. tried patching USB restrict with MinaUSB but nothing works so I decided to try it in terminal and and I got error communicating with the device.

iOS 13.0
iPhone X 
Error -20 on checkra1n ( which is know to be USB restriction)

But I basically can't communicate with the device.. any CMD to do this or other means to communicate with this device using terminal?? 

Mac OS: Mojave 
![Uploading 91499550...]![Uploading 91499571...]