libimobiledevice/libimobiledevice #827: iOS13. Linux. ideviceimagemounter -l exits with 141. Problem identified.

JonGabilondoAngulo, 2019-08-12:
I can see the problem only in Linux and ios13. 
The commands does the work OK but exits with 141. It means it exists due to SIGPIPE. 
And it is caused by 
`mobile_image_mounter_free(mim);`
which ends up in 
`idevice_disconnect`

This operation is closing pipes that are going to be written, causing a SIGPIPE. 
Problem identified, but not solved yet.

maznikoff, 2019-09-18:
The same issues with another tools on linux. Almost all the tools from libimobiledevice package exit with 141.

JonGabilondoAngulo, 2019-09-19:
It requires a deeper solution but by now I solved my situation  capturing the SIGPIPE.

#ifdef __linux__
static void on_signal(int sig)
{
    //  SIGPIPE captured to avoid the 141 exit value error.
}
#endif

..
main()
{
    #ifdef __linux__
        signal(SIGPIPE, on_signal);
    #endif

nikias, 2019-09-28:
Fixed with recent commit to ignore SIGPIPE.

closed 2019-09-28 by nikias