libimobiledevice/libimobiledevice #828: [ifuse] Invalid device UDID

R-a-l-f, 2019-08-16:
Hi, an iPad 2 (iOS 9.3.5) provides the desired ID, but an iPad Pro 3rd
generation (Model MTFR2FD/A, iOS 12.4) provides an invalid ID.

I don't know if it's related to [issue #770](https://github.com/libimobiledevice/libimobiledevice/issues/770 .)
`
[rocketmouse@archlinux ~]$ lsb_release -d
Description:	Arch Linux
[rocketmouse@archlinux ~]$ pacman -Qi libimobiledevice | grep -eVe -eBu
Version         : 1.2.0+87+g92c5462-1
Build Date      : Mon 19 Nov 2018 12:03:18 AM CET
[rocketmouse@archlinux ~]$ pacman -Qi ideviceinstaller-git | grep -eVe -eBu
Version         : 1.1.0.r24.g46e6e2a-1
Build Date      : Sat 25 May 2019 01:53:19 PM CEST
[rocketmouse@archlinux ~]$ pacman -Qi ifuse | grep -eVe -eBu
Version         : 1.1.3-6
Build Date      : Mon 04 Mar 2019 07:10:04 PM CET
[rocketmouse@archlinux ~]$ idevice_id -l
12c6f0748b9d3344303a0b5d775136ac4d4a7341
00008027-0014713A2122002E
[rocketmouse@archlinux ~]$ ifuse -u 00008027-0014713A2122002E ipad/ 
Invalid device UDID specified, length needs to be 40 characters
`
Regards,
Ralf

LeepengX, 2019-08-16:
waiting for answer

nikias, 2019-08-16:
Answer is short and simple: You are using an old version of ifuse. This has been fixed since October 1, 2018 with this commit: https://github.com/libimobiledevice/ifuse/commit/3b00243682962fec4f78f03d96cdded730cae38b.

nikias, 2019-08-16:
And obviously you have to compile from source to make sure you get the latest code.

R-a-l-f, 2019-08-16:
Note, all packages were build after October 2019, see my bug report. Unfortunately building libimobiledevice-git from the Linux distro's user repository fails at the moment, at least due to a dependency that doesn't build and I don't have the time right now to search for a workaround.

So ifuse version 1.1.3, build date Mar 2019 from the official repository of the rolling Linux release I'm using, doesn't contain the commit from October 2018?

If this is for sure, I'll report it to the distro's bug tracker. Please confirm. IMO it isn't obviously.

nikias, 2019-08-16:
@R-a-l-f I saw the dates but the error message you get would definitely not show if the commit was included.

R-a-l-f, 2019-08-16:
Thank you, in the meantime I noticed it and while you replied, I reported the bug via the Arch Linux bugtracker.

R-a-l-f, 2019-08-16:
Hi,

for those using Arch Linux a workaround:

`$ svn checkout --depth=empty svn://svn.archlinux.org/community`
`$ cd community/`
`$ svn update ifuse`
`$ cd ifuse/repos/community-x86_64/`
`$ cp PKGBUILD PKGBUILD.old`

Edit the PKGBUILD:

`$ diff PKGBUILD PKGBUILD.old`
`7,8c7,8`
`< pkgver=1.1.3.r6.ge75d32c`
`< pkgrel=1`
`---`
`> pkgver=1.1.3`
`> pkgrel=6`
`14,21c14,16`
`< source=("git+https://git.libimobiledevice.org/ifuse.git")`
`<` 
`< sha256sums=('SKIP')`
`< `
`< pkgver() {`
`<   cd "$pkgname"`
`<   git describe --long --tags | sed 's/\([^-]*-g\)/r\1/;s/-/./g'`
`< }`
`---`
`> source=("https://cgit.libimobiledevice.org/ifuse.git/snapshot/$pkgname-$pkgver.tar.bz2")`
`> md5sums=('30740edd3ea13b4361b3f1c09c6291fc')`
`> sha256sums=('ab42b61ed380eb9c44772ba9c5e38420cfc4bd112155e50181d9fd6fd5358852')`
`24c19`
`<   cd "$pkgname"`
`---`
`>   cd "$pkgname-$pkgver"`
`31c26`
`<   cd "$pkgname"`
`---`
`>   cd "$pkgname-$pkgver"`

Build and install the package (the path to the package might be different):

`$ makepkg -s`
`$ sudo pacman -U /var/cache/aur/ifuse-1.1.3.r6.ge75d32c-1-x86_64.pkg.tar.xz`

Done:

`$ mkdir ipad`
`$ idevice_id -l`
`00008027-0014713A2122002E`
`$ ifuse -u $(idevice_id -l) ipad/`
`$ ls -hAl ipad/DCIM/103APPLE/ | tail -2`
`-rw-r--r-- 1 rocketmouse rocketmouse 976K Jun 19 08:35 IMG_3179.JPG`
`-rw-r--r-- 1 rocketmouse rocketmouse 825K Jun 19 08:40 IMG_3190.HEIC`

Regards,
Ralf

mfowlerfinn, 2020-04-23:
Just ran into this today while using brew. It would be helpful to mark the new commit as a new release, since brew builds from the latest release and not the edge version...

For those still stuck:
`brew uninstall --ignore-dependencies ifuse`
`brew install --HEAD ifuse`

Works great now; Thanks!!