libimobiledevice/libimobiledevice #829: [off-topic] Not a bug. Request for a user mailing-list or forum

R-a-l-f, 2019-08-18:
Hi,

is there a forum or mailing list where I could get help with using libimobiledevice?

I try to write a script, to help people, who are not familiar with command line, but I need some help myself.

Mails sent to [libimobiledevice-devel](https://lists.libimobiledevice.org/mailman/listinfo/libimobiledevice-devel) don't came through the list and regarding the archive there was no traffic since December 2017, let alone that it isn't a user mailing-list.

IRC chat is no option for me.

Regards,
Ralf

nikias, 2019-08-18:
What is the purpose of the script?

R-a-l-f, 2019-08-18:
Hi,

the purpose is to share data with a Linux computer and to make this available for users who aren't familiar with command line and of course for what reason ever, won't or can't install e.g. gvfs. I for example don't install gvfs, because it makes green drives spin up, right after they go asleep, so the drives spin down and up again and again, until they are broken. After removing gvfs the drives stay asleep until the user touches the drive and they nearly live forever. To make a long story short, this is the message I tried to sent to the available mailing-list [1].

The text is wrapped at 72 chars, the script isn't wrapped in the original mail, neither is the PS, however, I removed the script and provide it as an attachment to this reply and removed the complete PS without adding it as an attachment.

-------- Forwarded Message --------
From: Ralf Mardorf
To: libimobiledevice-devel@libimobiledevice.org
Subject: AFC service 'com.apple.mobile.house_arrest' issue
Date: Sun, 18 Aug 2019 15:29:12 +0200

Hi,

since there's no user list, it might be ok, to sent a request to this
list.

Writing a script that could be used by people who aren't familiar with
command line fails. The goal is, that the script works by a GUI launcher
and does everything automagically.

Mounting iDevices to access the DCIM folder is no problem, but mounting
apps to access the documents fails for most, but not for all apps, if I
try to mount them by a list containing all apps, even those with the
'UIFileSharingEnabled' key not set.

An example that is always mountable is 'com.moogmusic.animoog'.
An example that can be mounted manually, but fails when trying to mount
it by the briefly described script is 'com.steinberg.cubasis', since it
fails with "Failed to start AFC service 'com.apple.mobile.house_arrest'
on the device."

However, if I unmount the iDevice/s that where mounted by the script,
it's possible to mount 'com.steinberg.cubasis' and other apps again and
to access their documents. This works without the need to
shutdown/restart or even to re-plug the USB connectors.

I hoped that a delay when automatically trying to mount one app after
the other could workaround the issue, but it doesn't.

I suspect that trying to mount some apps that don't allow file sharing
are the culprits. But perhaps it's something else.

Is there a way to get a list of all apps that support file sharing and
to automatically mount those apps, without running into the
'house_arrest' issue?

Tested with an iPad 2, iOS 9.3.5 and an iPad Pro 3rd gen, iOS 12.4 on
Arch Linux with

$ pacman -Q libimobiledevice ifuse libplist libusbmuxd usbmuxd ideviceinstaller-git
libimobiledevice 1.2.0+87+g92c5462-1
ifuse 1.1.3.r6.ge75d32c-1
libplist 2.0.0+11+gec9ba8b-2
libusbmuxd 1.0.10+29+g9db5747-1
usbmuxd 1.1.0+48+g1cc8b34-3
ideviceinstaller-git 1.1.0.r24.g46e6e2a-1

The (draft of a rough) script I used for testing:

$ cat imnt
[script removed and attached:]
[imnt.zip](https://github.com/libimobiledevice/libimobiledevice/files/3513316/imnt.zip)

Regards,
Ralf

PS: I wonder if 'ideviceinstaller -l -o xml' could be useful?

$ ideviceinstaller -l -o xml | grep UIFileSharingEnabled -B5 -A5 | grep [snip]]