libimobiledevice/libimobiledevice #832: Document Sharing fails not transparently when device locked

smidm, 2019-08-27:
Accessing application data using Document Sharing fails with "operation not permitted" while the device is locked. This is not transparent to the user and also not documented anywhere (ifuse manpage?).

```
# unlock the device
$ ifuse --documents com.liaoyuan.huo.FKCamera /media/FKCamera
$ rsync -P /media/FKCamera/x.mov  ~/

x.mov
  9,991,658,664 100%   10.73MB/s    0:14:48 (xfr#1, to-chk=0/1)

# lock the device
$ rsync -P /media/FKCamera/x.mov  ~/   

rsync: send_files failed to open "/media/x.mov": Operation not permitted (1)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1189) [sender=3.1.3]

```

This affects also the gvfs interface.

Device: iPhone 6
System: Fedora 30

mexmer, 2019-08-28:
this is how it works. Document sharing internaly works similar way to AFC, there is no "extended code" like device locked. it returns access denied, that is all. and libimobile does not know the reason, why access is denied.

mind this behavior is actually not caused by "locking display", but by device file encryption.

smidm, 2019-08-28:
The signalling is unfortunate. It would be interesting to know, how iTunes File Sharing is dealing with the problem.

I would suggest that `ifuse --documents` prints a warning message and that the issue would be mentioned in the `ifuse` manpage. It took me few hours to figure out the problem. This is completely unnecessary.

mexmer, 2019-08-28:
itunes only say access denied, nothing more.