libimobiledevice/libimobiledevice #833: Problem with openssl 1.1.1c

mexmer, 2019-08-28:
Because i want to look at some communication details, i decided to compile libimobile on my debian from scratch.

i compiled usbmuxd, libplist, libusbmuxd, libimobiledevice from sources.
idevicepair works without issues, usbmuxd seems to pair succesfully, but when i run ideviceinfo -d i get in debug output

```
idevice_connection_enable_ssl(): Error in SSL_do_handshake: SSL_ERROR_SSL
```

when i use version shipped with debian, it does work without problems, but because it has no debug code enabled, i can't see communication.

any idea where should i look (i did run usbmuxd as root, and running ideviceinfo as root also)

i specified openssl version in title, because that's only thing that comes to me mind, that might have any relation to this issue.

i use latest code from master.

mexmer, 2019-08-29:
when i removed patch with SSL_CTX_set_option it started working. i would suggest to remove this change, and use my proposed "fix" instead.

mexmer, 2019-08-29:
oh i forgot to mention one thing, since SSL is required for lockdown connection change ( a9ccd31 ) also breaks USBmuxd.

nikias, 2019-08-29:
@mexmer thanks for pointing that out. `SSL_CTX_set_max_proto_version` is only introduced with OpenSSL 1.1.0 so we can't change the code completely. Also why SSL3_VERSION and not TLS1_VERSION (for all iOS versions)?
I came up with this patch instead, that *should* work and also respects libressl versions:
```
diff --git a/src/idevice.c b/src/idevice.c
index 382e9d2..380d95c 100644
--- a/src/idevice.c
+++ b/src/idevice.c
@@ -789,6 +789,8 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_enable_ssl(idevice_conne
                return ret;
        }

+#if OPENSSL_VERSION_NUMBER < 0x10100002L || \
+       (defined(LIBRESSL_VERSION_NUMBER) && (LIBRESSL_VERSION_NUMBER < 0x2060000fL))
        /* force use of TLSv1 */
 #ifdef SSL_OP_NO_TLSv1_1
        int opts = SSL_OP_NO_TLSv1_1;
@@ -800,6 +802,9 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_enable_ssl(idevice_conne
 #endif
        SSL_CTX_set_options(ssl_ctx, SSL_OP_ALL | opts);
 #endif
+#else
+       SSL_CTX_set_max_proto_version(ssl_ctx, TLS1_VERSION);
+#endif

        BIO* membp;
        X509* rootCert = NULL;
```

mexmer, 2019-08-29:
i had issue with TLS1 negotiation on IOS v3, i'm using 1.1.1k version on windows. but i will let your know. if this works under debian, might be some issue with windows library version.

as for passing around product version, i thought that was already done, since you need product version in preflight (different behavior for ios < 7), so you can store it into idevice struct from usbmuxd_connect, and then pass it idevice_connection struct.

at least that's how i do it on windows, because there are some subtle changes in difference service behaviors and data access, so i pass version whenever i make new connection, after device is setup.

mexmer, 2019-08-29:
OK, just finished test, your patch works for ios 12 and 13, doesn't work for ios 5.1 and 3.1.2 device

mexmer, 2019-08-29:
correction, ios 13 is broken also, seems it requires higher TLS version, most likely 1.2

mexmer, 2019-08-29:
you will need to pass ios version around, otherwise you will not be able to keep compatibility with old ios.

nikias, 2019-08-29:
Looking into making the changes now.

mexmer, 2019-08-29:
i will do some experiments, can you check on your side?
i will try different changes with different devices.

but need to recompile usbmuxd/libimobile/libusbmux everytime, just to be safe 

i have device with ios 12.4, 13.0 (latest public beta), ios 5.1.1 and ios 3.1.2, so i'm checking it with them.

on windows my code (not libimobile) looks like this 
```
	SSL_CTX * Context = NULL;
	Context = SSL_CTX_new(TLS_method());
	if (SSL_CTX_set_min_proto_version(Context, iOSVer > 11 ? TLS1_1_VERSION : SSL3_VERSION) == 1)
	{
		if (iOSVer < 9)
			if (SSL_CTX_set_max_proto_version(Context, SSL3_VERSION) == 0)
				return ERROR_INVALID_PARAMETER;
	}
```
min version setting is most likely not needed.

mexmer, 2019-08-29:
OK finished my testing

min - not set
max - not set
ios 5.1.1
ios 4.2.1 
ios 3.1.2 does not work

min and max set to TLS1
ios 3.1.2 doesn't work

min and max set to SSL3
ios 12.4 
ios 13.0 doesn't work

min not set and max set to TLS1_1
ios 5.1.1
ios 4.2.1 
ios 3.1.2 
ios 12.4 
ios 13.0 doesn't work

min and max set to TLS 1_1
ios 5.1.1
ios 4.2.1 
ios 3.1.2 doesn't work

min not set and max set to TLS 1_2
ios 5.1.1
ios 4.2.1 
ios 3.1.2 doesn't work

mexmer, 2019-08-29:
honestly i have no idea how openssl (or apple) is handling this. i know there were issues with negotiating SSL/TLS version. there were also some patches to prevent "downgrading" protocol (there is CVE for that).

all i can say, that this behavior needs to be taken into account if you want to maintain compatibility across multiple iOS versions.

while TLS1 seems to have best compatibility (only exception is ios 3) in future (And i expect it might come with upcoming TLS 1.3 - unless they rename it to 2.0) they might even refuse backward protocol compatibility.

nikias, 2019-08-29:
Thanks for testing. So you are saying, with min and max set to TLS1_1 it will work with iOS >= 4 and including 13 ? Have you tried min and max with TLS1.0 to see where it fails? And also, for iOS 3, do we need SSL3_VERSION for min and max?

mexmer, 2019-08-29:
yes io3 worked only with MIN/MAX 3,
for other ioses, if i did set only MAX TLS1 none worked, with MIN/MAX TLS1 all worked up to ios13
with MIN TLS1 (max not set) only 12 and 13 

i did not mention in list above, but with MIN 3, MAX TLS 1 result was same, like with MIN TLS1/MAX TLS1, eg. OS 3 didn't work, everything else did work.

i will try to find another iOS 3 device, because this one is jailbroken, with sideloaded cydia and ssh server, but for now, it's only iOS 3 device i have.

mexmer, 2019-08-29:
as for where it fails. it always fails on handshake. which means, preflight fails also when it try to connect to lockdown. pair comes trough (because it's unencrypted), but that's all. i can send you logs from linux if needed.

nikias, 2019-08-29:
That's ok, I don't need logs. It fails because of SSL version, I debugged it the other day. I also don't have an iOS 3.x device handy but it seems we need to set it to SSL3_VERSION for iOS < 4 and for everything else we can set to TLS1_VERSION min+max. We also need to figure out how to handle this for older OpenSSL/LibreSSL versions that don't have SSL_CTX_set_min/max_proto_version functions. LibreSSL 2.2.7 - which is the default on recent MacOS - for example doesn't have them.

mexmer, 2019-08-29:
unfortunately i'm not working on mac, so i have no idea, how to do it there, but maybe it works different so it does handle it properly?

i will try next week to as coleague to bring his macbook, and hopefully test it.

nikias, 2019-08-29:
it's ok that will be fine, I just need to have things tested with different iOS versions, I don't have a 3.x or 4.x or 5.x device handy :)

mexmer, 2019-09-05:
let me know, when there is ready version for ios and i can test it.

mexmer, 2019-09-11:
ok, this is now working with openssl 1.1 on debian, also with openssl 1.0.2 on OS X

closed 2019-09-11 by mexmer