libimobiledevice/libimobiledevice #835: idevicebackup2 change password on ios13

mexmer, 2019-08-29:
@nikias on iOS 13 changing or settings backup password requires PIN confirmation. can you add message to changepw command (ios 13 only), that user should confirm change on phone?

because now it just shows started service, then negotiated protocol, then nothing happens, because it's wating for device response, which doesn't come until user confirms change by his passcode.

this might be very confusing if people are not looking at the phone.

btw. itunes 12.9 (current version on windows), doesn't inform about passcode confirmation requirement either, i hope apple will fix that, before official ios13 release.

nikias, 2019-09-27:
See commit 1faa16c912dd5eb4e9305b25b13407de195ad103.

closed 2019-09-27 by nikias

mexmer, 2019-09-30:
oh didn't know, you can ask device, if it has password set. nice

nikias, 2019-09-30:
Yes, I figured relying on PasswordProtected would only work if the device is on lockscreen, so I came up with this :)

mexmer, 2019-09-30:
can you query that with locked screen?

nikias, 2019-09-30:
Yes, if the device is paired you can.