libimobiledevice/libimobiledevice #838: idevice_connect() returns IDEVICE_E_UNKNOWN_ERROR on trying to connect to the second device

kakyoism, 2019-09-05:
I'm trying to connect a Windows machine to two iOS devices at the same time. The first iOS device works, but the second one fails to connect using a different port. The code for setting up connection looks like this

```c++
        idevice_error_t res = idevice_new(&m_device, m_id.c_str());
	if (res != IDEVICE_E_SUCCESS) {
		spdlog::error("Error creating iOS Guest {}: {}. Aborted.", m_id.c_str(), res);
		return My_ErrorCreatingiOSDevice;
	}

	// Set up connetion.
	const int nRetries = 10;
	uint16_t port = 2345;
	res = IDEVICE_E_UNKNOWN_ERROR;
	for (auto t = 0; res != IDEVICE_E_SUCCESS && t < nRetries; ++t, ++port) {		
		res = idevice_connect(m_device, port, &m_connection);
		if (res != IDEVICE_E_SUCCESS) {
			spdlog::error("Error setting up connection for iOS Guest {} at Port {}: {}. Will retry next port {}.", m_id.c_str(), port, res, port+1);
			Sleep(1000);
		}
		else {
			spdlog::debug("SUCCESS: CommandTunnel for iOS Guest {} is set up at Port {}.", m_id.c_str(), port);
			break;
		}
	}
	if (res != IDEVICE_E_SUCCESS)
		return My_ErrorConnection;
```
Is it that libimobiledevice only supports one device at a time or am I missing something obvious?
Is it a limitation

nikias, 2019-09-05:
No, it's not a limitation and it should just work. What's the returned error code `res` ?

kakyoism, 2019-09-05:
@nikias It's `IDEVICE_E_UNKNOWN_ERROR` as my title says. I traced into `libusbmuxd` and found out more errors:

```
libimobiledevice\src\idevice.c:278 idevice_connect(): ERROR: Connecting to usbmuxd failed: -1 (Operation not permitted)
```

Unfortunately, my VS2017's breakpoints don't work inside `usbmuxd` after attaching to the process.
But it's gotta be in 
```
int socket_connect(const char *addr, uint16_t port)
```
```
libusbmuxd\common\socket.c
```

kakyoism, 2019-09-05:
OK, further trying shows that it's the second iPhone itself that 's causing trouble. It's an iPhone7 of Model `MNAK2LL/A` running latest `iOS 12.4`. I can't figure out where it went wrong, but experience tells me it might some permission issue that requires user to enable certain Xcode app plist options or hidden Settings on the phone...

I don't know. Any tips?

kakyoism, 2019-09-05:
Update: I was finally able to trace into `libusbmuxd` (previously I had an outdated .dll). The error with the second phone happens in `usbmuxd_connect()` call in `libusbmuxd.c`, where `usbmuxd_get_result` gives me a `res` code `3`, which seems to refer to `RESULT_CONNREFUSED = 3`. So more and more like a permission issue on the phone. 

Funny thing is I've gathered 3 iPhones, one iPad Mini, and and iPad Air2. Only one iPhone6 works which runs `iOS 12.4`.  The iPhone7 and iPad Air run `iOS 12.4.1`, another iPhone6 runs `iOS 12.3.1`. 

I still couldn't figure out what's different with the setup.

mexmer, 2019-09-05:
@kakyoism what exactly you trying to do, a why you setting port to 2345?

you understand, that you need first to connect to lockdown port, then start service, which will return you service port, then you connect to that port?

lockdown port has same value on all phones.
you do not select different port for different device, connection is multiplexed by usbmuxd.

you get connection refused, because on port you have specified, there is no service.

nikias, 2019-09-05:
@mexmer I think he might have a listening port open on some app that is running on the device. It's basically what iproxy allows you to do.

mexmer, 2019-09-05:
yes, but when you get connection refused, it means that device refuses connection, eg. nothing is running on that port.

nikias, 2019-09-05:
@mexmer yes of course. Thing is we both don't understand what the port is for and how the setup looks like :)

kakyoism, 2019-09-06:
@mexmer @nikias Thanks for asking. My mobile app listens to  Port `2345`. I'm new to this code-base and thought `2345` was the port on the PC-side server, but it's the mobile server that's listening to that port. But shouldn't a socket server be able to accept multiple connections on that same port?

kakyoism, 2019-09-06:
Finally solved it. It really was a coding error on our side. The hard-coded 2345 was temporarily set to another one by someone else and suddenly all phones got affected except one that was still running the old version of the app. So the moral lesson is to always trust the obvious explanation until enough evidence says otherwise: wrong socket configuration. 

But since I'm new to this,  I'd still love to clarify my other questions with you guys:

1. By lockdown port, do you mean the #define USBMUXD_SOCKET_PORT 27015 found in usbmuxd-proto.h?
2. Could I open several serves on the mobile side, each listening to a different port, and have the same PC/Mac connect to them individually, i..e, multiple communication channels from a single client over the same `lockdown port`?

Thanks a bunch!

nikias, 2019-09-06:
Ok that explains everything now :)
Yes you can have multiple listening ports on the device and write code that does idevice_connect for each of the ports. Ignore the lockdown port, that's something else (required to start services on device) and also the usbmuxd port (internally handled by libusbmuxd).

mexmer, 2019-09-06:
as nikias mentioned, since your service is already running on phone, you don't need to connect to lockdown.

all devices can use same port, because when you call idevice_connect important is, that each device has own identifier. idevice_connect instructs usbmux server to make connection to device (specified by id) to port opened on device, then it will hand back socket to you, which you use for communication.

usbmux does multiplexing (and proxying) connection for you.

kakyoism, 2019-09-07:
@nikias @mexmer Thank you both for your patience! It really helped. I'll close it now.

closed 2019-09-07 by kakyoism