libimobiledevice/libimobiledevice #841: [idevicerestore2] Options `--system` and `--settings` have unclear meaning

intelfx, 2019-09-09:
`idevicerestore2 restore` has certain flags whose meaning is not immediately obvious from end-user perspective:

```
  restore       restore last backup to the device
    --system            restore system files, too.
<...>
    --settings          restore device settings from the backup.
<...>
```

Specifically, when I tried to restore a full iOS 12 backup after device restore, default options gave an incomplete restore (the phone didn't remember my Apple ID, general settings and app icon layout at least). Which combination of restore options will give a result equivalent to an iTunes restore?

Probably makes sense to document the combination, too (or make it the default and add inverse options).

It is not clear which combination of restore options will give a result equivalent to an iTunes restore.

nikias, 2019-09-09:
Yes indeed the options are confusing. Restore with what iTunes is doing would be with 
`--system --settings`.

MisutaaAsriel, 2026-01-30:
What do the options mean though.

What data is "system" and what data is "settings".

I imagine Settings *should* be obvious (device settings, configuration), but what is "system"? Is that app data? Is that everything?