libimobiledevice/libimobiledevice #842: If the model is repaired and replaced with a new part, how to check the previous value

slin1972, 2019-09-09:
Such as
After the repair
WIFI-Address  from  aa:bb:cc:dd:ee change to dd:bb:dd:cc
How do I get previous(old) wifi-address.

Looking forward to your reply

nikias, 2019-09-19:
You can only get the old WiFi mac address before the repair. Afterwards there's no way to find the old one.

slin1972, 2019-09-22:
But I have seen some software that can read the information before the phone is repaired. To determine if it has been repaired

so  
I really want to know how does it get the factory value

[image]

nikias, 2019-09-22:
If you get me the software I may have a look.

slin1972, 2019-09-23:
https://www.i4.cn
This is its website. You can download it there.
[image]

slin1972, 2019-09-23:
[image]
[image]

Blefish, 2019-09-23:
The same interface is used by 3uTools, I don't know which one is the original.

slin1972, 2019-09-23:
[image]

slin1972, 2019-09-23:
The data of i4 is more accurate.

slin1972, 2019-09-23:
Can you see how it got the factory value?

mexmer, 2019-09-23:
question is, if this are really genuine values. like on first screen, it shows even different model.

slin1972, 2019-10-08:
The value is correct.

slin1972, 2019-10-08:
@nikias

slin1972, 2019-10-08:
I found that it automatically installed a software on my phone
[image]

nikias, 2019-10-08:
Well the app can't run if you don't trust the developer. Installing is always possible.

closed 2020-03-02 by slin1972

tangwanchao, 2020-08-03:
@slin1972 @nikias 
So how do you get the factory data?

msk521, 2021-12-14:
any body can  get the factory data?

truonggiang0710, 2023-08-09:
@slin1972 @tangwanchao @msk521 did you have any solution to do it?

zuodd, 2024-11-25:
I dont think this post can be closed. No one give the answer. @mexmer @nikias @slin1972