libimobiledevice/libimobiledevice #845: Update on network support (Wifi Sync)

Soutcast, 2019-09-16:
I was just wondering whether anyone has any updates on how the development for communication with devices over wifi is going? Any input would be appreciated.

nikias, 2019-09-18:
It's coming. Working on cleanup of the code as we speak.

Soutcast, 2019-09-18:
@nikias, thanks for the update. Can you give any estimated time of when you will release it?

JamesBedwell, 2019-10-18:
@nikias do you have an update? I'm eagerly awaiting wifi support :)

aovestdipaperino, 2019-10-25:
Ping @nikias 
Is there any fork/branch that can be used even if the code is not perfectly cleaned up?

vlar, 2019-11-16:
Still eagerly waiting for the feature even 1 year after ;) I would also take a non cleaned code gladly

AlexShimmy, 2019-12-15:
@Soutcast @JamesBedwell @aovestdipaperino @vlar What tools would you like to use over the network?

tech234a, 2020-01-04:
I personally would like to install apps over the network. @AlexShimmy

Quentin-M, 2020-01-09:
@AlexShimmy Backups!

Soutcast, 2020-01-09:
@AlexShimmy Backups, App Installation, Screenshots

JamesBedwell, 2020-01-09:
Backups, and I also want to attempt to build a cross-device photo sync solution.

aovestdipaperino, 2020-01-11:
@AlexShimmy mostly installs.

alexw1982, 2020-01-25:
Backups

vlar, 2020-01-25:
The main reason for me is battery charging state for home automation. (if phone is charging or not) 
And then Backups.

chmiiller, 2023-02-24:
would love to measure battery consumption while running some apps