libimobiledevice/libimobiledevice #848: Backup on iOS 13 hangs at Sending 'XXXUDIDHEREXXX/Manifest.plist'

libreminded, 2019-09-24:
Is it only me or is there a Problem with idevicebackup2 on iOS 13?

I could always backup my iPhone 8 up to iOS 12.4.1 without any problems.
After upgrading to iOS 13.1 idevicabackup2 stays at Sending 'XXXUDIDHEREXXX/Manifest.plist' forever. I think it can't be my libimobildevice installation because I CAN backup my iPad on iOS 12.4.1 unproblematically. 

Hardware:
Thinkpad running Debian 10 with libimobiledevice-utils Package (1.2.1~git20181030.92c5462-1)
iPhone 8 (MQ6G2ZD/A)

mexmer, 2019-09-24:
you need compile from source. packages in debian do not work with ios13. you even have date of sourcecode in package name, obviously code from 2018 will not work with system, that was released few days ago.

libreminded, 2019-09-24:
@mexmer Thx for your reply. I'll try the compiled version. Yeah I know the debian packaged version is pretty outdated, but the official website states version 1.2.0 as the current one.

corsac-s, 2019-09-25:
Hi, I'm the Debian maintainer for libimobiledevice and friends. I've packaged a new snapshot which is currently sitting in NEW queue and should be available in unstable soon.

I'll try to ask for a stable upload as well but since it's not a small diff it might be an issue. @libreminded is there a specific list of commits (on libimobiledevice and libusbmuxd) which I could cherry-pick?

corsac-s, 2019-09-25:
I just tried with libimobiledevice `a9e69b1` and libusbmuxd `b097ea3` on an iOS 13 device and the backup never finishes, with;

```
[==================================================] 100% Finished
Sending '<UUID>/Status.plist' (189 Bytes)
Sending '<UUID>/Manifest.plist' (83.8 KB)
Device is not ready yet. Going to try again in 2 seconds...
```
last line looping over and over.

nikias, 2019-09-27:
I just tried and it works fine for me. iPhone 11 Pro, iOS 13.1 - So from what I see it doesn't send Manifest.db which should be right after the Manifest.plist. Can you run idevicesyslog on another terminal and see what it complains about?

corsac-s, 2019-09-28:
I'm trying on an iPhone SE, iOS 13.1. I've tried idevicesyslog but it's really noisy, I've filtered (grep) on BackupAgent2 but I'm unsure if it's ok:
```
Sep 28 13:49:19 slab BackupAgent2(libMobileGestalt.dylib)[1511] <Notice>: elided platform fast path for key: <some identifier>
Sep 28 13:49:19 slab BackupAgent2(libMobileGestalt.dylib)[1511] <Notice>: elided platform fast path for key: <some identifier>
Sep 28 13:49:19 slab BackupAgent2(libMobileGestalt.dylib)[1511] <Notice>: elided platform fast path for key: <some identifier>
Sep 28 13:49:19 slab BackupAgent2[1511] <Notice>: Device UDID: <private>
Sep 28 13:49:19 slab BackupAgent2(liblockdown.dylib)[1511] <Notice>: unlock_with_escrow: <private>
Sep 28 13:49:19 slab BackupAgent2(MobileKeyBag)[1511] <Notice>: MKBKeyBagUnlock: unlocked bag 1
Sep 28 13:49:20 slab BackupAgent2(MobileBackup)[1511] <Notice>: Loading system domains from /System/Library/Backup/Domains.plist
Sep 28 13:49:20 slab BackupAgent2[1511] <Notice>: Action: Backup
Sep 28 13:49:20 slab BackupAgent2[1511] <Notice>: Starting backup
Sep 28 13:49:20 slab BackupAgent2[1511] <Notice>: Starting preflight
Sep 28 13:49:20 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:20 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
Sep 28 13:49:21 slab BackupAgent2(libsystem_containermanager.dylib)[1511] <Notice>: Requesting group container lookup; identifier = <private>, euid = <private>, uid = <private>, personaid = <private>, type = <private>, name = <private>
[... cut ...]
Sep 28 13:49:22 slab BackupAgent2[1511] <Notice>: Notified MBSiriPlugin of startingBackupWithEngine: in 0.000s
Sep 28 13:49:22 slab BackupAgent2[1511] <Notice>: Notified MBKeyboardPlugin of startingBackupWithEngine: in 0.000s
Sep 28 13:49:22 slab BackupAgent2[1511] <Notice>: Backup status: <private>
Sep 28 13:49:40 slab BackupAgent2(iTunesCloud)[1511] <Notice>: Battery level changed. batteryLevel=0.90
```
Then  nothing happens, while idevicebackup2 is looping on the _Going to try again in 2 seconds_ message

corsac-s, 2019-09-28:
I tried to remove `Status.plist` and `Manifest.plist` from the local backup, and start it again. The backup process does some stuff (backing files up) but in the end I'm back to the:

```
Sending '2c18f2ee4a53d2ca69482826c597b4ee7d40c849/Status.plist' (189 Bytes)
Sending '2c18f2ee4a53d2ca69482826c597b4ee7d40c849/Manifest.plist' (84.0 KB)
Device is not ready yet. Going to try again in 2 seconds...
```
step.

nikias, 2019-09-28:
Can you go to Settings -> Privacy -> Analytics -> Analytics Data and check if there's any entry for BackupAgent2 ?

corsac-s, 2019-09-29:
I have two entries but they're from 2018 so it's unlikely to be it, I guess?

nikias, 2019-09-29:
Maybe you have too many crash logs and it won't create new ones. You can run it with idevicesyslog again and filter for Crash or BackupAgent2:
```
idevicesyslog |grep -E "Crash|BackupAgent2"
```
Also I recently updated the code so it can differentiate between a timeout condition ('Device is not ready yet') vs. an error condition (e.g. connection closed). Please run with a fresh build. It won't solve the problem but at least we'll understand if e.g. the connection died or something.

corsac-s, 2019-09-29:
I just tried `656b96c` fresh from the repository and it completes successfully. I'll try to package that version for Debian and maybe investigate which revision actually did the trick.

corsac-s, 2019-09-29:
The packaged version didn't work either. One difference between packaged version and the one locally built from the repository is the SSL backend.

The default is to use OpenSSL, which works fine, while in Debian we use the GnuTLS backend (likely for licensing reason) and it seems broken somehow with iOS 13.

nikias, 2019-09-29:
Ok so maybe it's a communication issue after all... (GnuTLS related). There were a few fixes for OpenSSL recently but those issue were actually during initial handshake and not while communication is in progress...

nikias, 2019-09-29:
Is there any chance you can package it with --enable-debug-code configured, and then run idevicebackup2 with `-d` ?

corsac-s, 2019-09-29:
Sure. Here's the end of the log (I can provide the full one if needed, but maybe by mail?)

```
Sending '<UDID>/Manifest.plist' (84.0 KB)
15:32:40 service.c:144 service_send(): sending 5 bytes
15:32:40 idevice.c:631 internal_ssl_write(): pre-send length = 37
15:32:40 idevice.c:636 internal_ssl_write(): post-send sent 37 bytes
15:32:40 service.c:144 service_send(): sending 32768 bytes
15:32:40 idevice.c:631 internal_ssl_write(): pre-send length = 16421
15:32:40 idevice.c:636 internal_ssl_write(): post-send sent 16421 bytes
15:32:40 service.c:147 service_send(): ERROR: sending to device failed.
15:32:40 service.c:144 service_send(): sending 4 bytes
15:32:40 idevice.c:631 internal_ssl_write(): pre-send length = 37
15:32:40 idevice.c:636 internal_ssl_write(): post-send sent 37 bytes
15:32:40 property_list_service.c:132 internal_plist_send(): sending 108 bytes
15:32:40 service.c:144 service_send(): sending 4 bytes
15:32:40 idevice.c:631 internal_ssl_write(): pre-send length = 37
15:32:40 idevice.c:636 internal_ssl_write(): post-send sent 37 bytes
15:32:40 service.c:144 service_send(): sending 108 bytes
15:32:40 idevice.c:631 internal_ssl_write(): pre-send length = 149
15:32:40 idevice.c:636 internal_ssl_write(): post-send sent 149 bytes
15:32:40 property_list_service.c:137 internal_plist_send(): sent 108 bytes
15:32:40 property_list_service.c:138 internal_plist_send(): printing 308 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
	<string>DLMessageStatusResponse</string>
	<integer>0</integer>
	<string>___EmptyParameterString___</string>
	<dict/>
</array>
</plist>
15:32:40 notification_proxy.c:48 np_lock(): Locked
15:32:40 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
15:32:40 notification_proxy.c:267 np_get_notification(): NotificationProxy: no notification received!
15:32:40 notification_proxy.c:59 np_unlock(): Unlocked
15:32:41 notification_proxy.c:48 np_lock(): Locked
15:32:42 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
```
After that it loops on the 4 last lines.

nikias, 2019-09-29:
Nah, we can see the issue here. "ERROR: sending to device failed."
This is the problem:
https://github.com/libimobiledevice/libimobiledevice/blob/master/src/idevice.c#L387-L398 :
```
#ifdef HAVE_OPENSSL
		int sent = SSL_write(connection->ssl_data->session, (const void*)data, (int)len);
		debug_info("SSL_write %d, sent %d", len, sent);
#else
		ssize_t sent = gnutls_record_send(connection->ssl_data->session, (void*)data, (size_t)len);
#endif
		if ((uint32_t)sent == (uint32_t)len) {
			*sent_bytes = sent;
			return IDEVICE_E_SUCCESS;
		}
		*sent_bytes = 0;
		return IDEVICE_E_SSL_ERROR;
```
As you can see in your log, it tries to send 32678 bytes, but GnuTLS only sends 16421 in the first chunk for this packet, but the above code only sees a success when `sent == len`. This code needs to be fixed, and then it will work.

nikias, 2019-09-29:
@corsac-s can you try this patch:
```diff
diff --git a/src/idevice.c b/src/idevice.c
index 90b531d..e60afd1 100644
--- a/src/idevice.c
+++ b/src/idevice.c
@@ -384,18 +384,25 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_send(idevice_connection_
        }

        if (connection->ssl_data) {
+               uint32_t sent = 0;
+               while (sent < len) {
 #ifdef HAVE_OPENSSL
-               int sent = SSL_write(connection->ssl_data->session, (const void*)data, (int)len);
-               debug_info("SSL_write %d, sent %d", len, sent);
+                       int s = SSL_write(connection->ssl_data->session, (const void*)data, (int)len);
 #else
-               ssize_t sent = gnutls_record_send(connection->ssl_data->session, (void*)data, (size_t)len);
+                       ssize_t s = gnutls_record_send(connection->ssl_data->session, (void*)data, (size_t)len);
 #endif
-               if ((uint32_t)sent == (uint32_t)len) {
-                       *sent_bytes = sent;
-                       return IDEVICE_E_SUCCESS;
+                       if (s < 0) {
+                               break;
+                       }
+                       sent += s;
                }
-               *sent_bytes = 0;
-               return IDEVICE_E_SSL_ERROR;
+               debug_info("SSL_write %d, sent %d", len, sent);
+               if (sent < len) {
+                       *sent_bytes = 0;
+                       return IDEVICE_E_SSL_ERROR;
+               }
+               *sent_bytes = sent;
+               return IDEVICE_E_SUCCESS;
        }
        return internal_connection_send(connection, data, len, sent_bytes);
 }
```

Shankar-mac, 2019-09-29:
For 13.1.1 & iPhone XR.. I am getting below error..

Sep 29 19:54:24 iPhone BackupAgent2[309] <Notice>: Verifying backup contents
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|SocketStreamHandler.c:_SocketLogCallback| ERROR: SocketRecv (data): error -555: Unknown error: -555
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|StreamHandler.c:_ReadFunc| ERROR: Error reading data: Can't read data
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|DeviceLinkConnection.c:_DLHandlerThreadReadMessage| ERROR: Could not receive message: '(null)'
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|SocketStreamHandler.c:_SocketLogCallback| ERROR: SocketRecv (data): error 22: Invalid argument
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|StreamHandler.c:_ReadFunc| ERROR: Error reading data: Can't read data
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|DeviceLinkConnection.c:_DLHandlerThreadReadMessage| ERROR: Could not receive message: '(null)'
Sep 29 19:54:35 iPhone BackupAgent2[309] <Notice>: Drive returned unexpected error code: 1
Sep 29 19:54:35 iPhone BackupAgent2[309] <Notice>: Finished backup in 18.998s
Sep 29 19:54:35 iPhone BackupAgent2[309] <Error>: Failed to backup: Error Domain=MBErrorDomain Code=103 UserInfo={NSLocalizedDescription=<private>, NSUnderlyingError=0x280110ab0 {Error Domain=MBErrorDomain Code=1 UserInfo={NSFilePath=<private>, NSLocalizedDescription=<private>}}}
Sep 29 19:54:35 iPhone BackupAgent2[309] <Notice>: Notified MBRestoreAppsPlugin of endedBackupWithEngine:error: in 0.000s
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|SocketStreamHandler.c:_SocketLogCallback| ERROR: SocketSend (data): error 22: Invalid argument
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|StreamHandler.c:_WriteFunc| ERROR: Error sending data: Can't send data
Sep 29 19:54:35 iPhone BackupAgent2[309] <Notice>: Notified MBLockdownPlugin of endedBackupWithEngine:error: in 0.000s
Sep 29 19:54:35 iPhone BackupAgent2(MobileDeviceLink)[309] <Notice>: 309:1831284736|DeviceLinkConnection.c:_DLSendBytesAndDisconnectOnError| ERROR: Could not send data: 'Error writing length of data'
Sep 29 19:54:35 iPhone powerd[36] <Notice>: Process BackupAgent2.309 Released NoIdleSleepAssertion "BackupAgent2" age:00:00:19  id:4295000518 [System: SysAct]
Sep 29 19:54:35 iPhone BackupAgent2[309] <Notice>: Notified MBAppleCarePlugin of endedBackupWithEngine:error: in 0.008s
Sep 29 19:54:35 iPhone BackupAgent2[309] <Notice>: Notified MBAggregateDictionaryPlugin of endedBackupWithEngine:error: in 0.000s
Sep 29 19:54:35 iPhone BackupAgent2[309] <Notice>: Unmounting APFS snapshot <private> at /.ba and removing it
Sep 29 19:54:35 iPhone BackupAgent2(MobileBackup)[309] <Notice>: Unmounted snapshot named <private> at: /.ba
Sep 29 19:54:35 iPhone BackupAgent2(MobileBackup)[309] <Notice>: Deleted snapshot: <private>
Sep 29 19:54:35 iPhone BackupAgent2[309] <Error>: Computer disconnected: Lost connection
Sep 29 19:54:35 iPhone BackupAgent2(MobileBackup)[309] <Notice>: Removing temporary directory: <private>
Sep 29 19:54:35 iPhone BackupAgent2(MobileBackup)[309] <Notice>: MBExit(1)
Sep 29 19:54:35 iPhone runningboardd(RunningBoard)[26] <Notice>: [executable<BackupAgent2>:309] Death sentinel fired!
Sep 29 19:54:35 iPhone runningboardd(RunningBoard)[26] <Notice>: Removing process: [executable<BackupAgent2>:309]
Sep 29 19:54:35 iPhone runningboardd(RunningBoard)[26] <Notice>: Removing assertions for terminated process: [executable<BackupAgent2>:309]
Sep 29 19:54:35 iPhone SpringBoard(FrontBoard)[55] <Notice>: [executable<BackupAgent2>:309]* Now flagged as pending exit for reason: process exited with context
Sep 29 19:54:35 iPhone SpringBoard(FrontBoard)[55] <Notice>: [executable<BackupAgent2>:309]* Process exited: <RBSProcessExitContext; unknown>.
Sep 29 19:54:35 iPhone SpringBoard(FrontBoard)[55] <Notice>: [executable<BackupAgent2>:309]* Setting process task state to: Not Running
Sep 29 19:54:35 iPhone SpringBoard(FrontBoard)[55] <Notice>: Removing: <FBProcess: 0x281372d80; executable<BackupAgent2>:309:309>
Sep 29 19:54:46 iPhone kernel(Sandbox)[0] <Notice>: Sandbox: 2 duplicate reports for BackupAgent2 deny(1) file-unmount

corsac-s, 2019-09-29:
> @corsac-s can you try this patch:
github apparently messes with tabs so I had to patch “manually”, but it doesn't work:

16:55:16 service.c:144 service_send(): sending 18438 bytes
16:55:16 idevice.c:640 internal_ssl_write(): pre-send length = 16421
16:55:16 idevice.c:645 internal_ssl_write(): post-send sent 16421 bytes
16:55:16 idevice.c:389 idevice_connection_send(): sent=16384, len=18438
16:55:16 idevice.c:640 internal_ssl_write(): pre-send length = 16421
16:55:16 idevice.c:645 internal_ssl_write(): post-send sent 16421 bytes
16:55:16 idevice.c:401 idevice_connection_send(): SSL_write 18438, sent 32768
Error: sent only 32768 of 18438 bytes

But I have a feeling there's an offset error here:
> ```diff
> diff --git a/src/idevice.c b/src/idevice.c
> index 90b531d..e60afd1 100644
> --- a/src/idevice.c
> +++ b/src/idevice.c
> @@ -384,18 +384,25 @@ LIBIMOBILEDEVICE_API idevice_error_t [...]
Shouldn't we start at data+s or something like that here?

nikias, 2019-09-29:
oooooh wait let me fix it

nikias, 2019-09-29:
Updated the patch in my comment above.

corsac-s, 2019-09-29:
Here's a proposal which _seems_ to work, but please double check

corsac-s, 2019-09-29:
> Updated the patch in my comment above.

Yes that seems to work

nikias, 2019-09-29:
> > Updated the patch in my comment above.
> 
> Yes that seems to work

Seems to work or does work? I mean theoretically it should work but I want real results :)

corsac-s, 2019-09-29:
> Seems to work or does work? I mean theoretically it should work but I want real results :)

It does work, the backup succeeds:

```Sending '<UDID>/Status.plist' (189 Bytes)
Sending '<UDID>/Manifest.plist' (84.0 KB)
Sending '<UDID>/Manifest.db' (54.9 MB)
Received 234 files from device.
Backup Successful.
```

nikias, 2019-09-29:
See 60823f9eeb3f09cd8083f613522d01e21d2f6d2d.

corsac-s, 2019-09-29:
Works fine. And the good news is that it applies fine against the version in Buster, and does seem to fix the problem there as well, so I should be able to request a stable upload and fix the issue for @libreminded 

Thanks!

closed 2019-09-30 by nikias

TravelTrader, 2019-12-13:
I have the same issue.
But I do not exactly understand what the solution was.

I use `idevicebackup2 backup <directory>` and it works well. But after the last lines there's no error but nothing else happens:

```
...
[==================================================] 100% Finished
Sending '<manycharacters>/Status.plist' (189 Bytes)
Sending '<manycharacters>/Manifest.plist' (121.4 KB)
```

What can I do to solve it?
Where do I have to enter the patch?
Thank you very much.

michalM-netblink, 2019-12-19:
> I have the same issue.
> But I do not exactly understand what the solution was.
>
> I use `idevicebackup2 backup <directory>` and it works well. But after the last lines there's no error but nothing else [...]

Did you manage to solve that issue?

michalM-netblink, 2019-12-19:
Did you manage to solve that issue?

TravelTrader, 2019-12-19:
Someone answered (I got an email) and now I can't see anything here.
But it's not solved yet.

corsac-s, 2019-12-21:
> What can I do to solve it?
> Where do I have to enter the patch?
> Thank you very much.

If you build from source, make sure you update to latest git. If you use packages provided by a Linux distribution, make sure they're up to date (for example Debian buster/stable needs 1.2.1~git20181030.92c5462-2 but unfortunately it's not yet on the mirrors)

bundabrg, 2019-12-26:
I compiled the latest git and can confirm the patch there. However I still have this same issue.

bundabrg, 2019-12-26:
Sorry, I can confirm it works. All my binaries linked to the library already installed on my system under /usr/lib/x86_64 instead of /usr/local/lib. Once I fixed that it works fine. Sorry for the spam.

TravelTrader, 2020-01-02:
After "some" difficulties I got the latest git installed with [this how-to](https://askubuntu.com/questions/598940/libimobiledevice-1-2-ios-8-support-for-ubuntu-14-04-trusty/712154#712154) (only `gnutls-dev` instead of `libgnutls-dev`).

But after running `idevicebackup2 backup <directory>` it again never finished...

beckietech, 2022-01-06:
hi @TravelTrader 

I'm new to see this issue, and I just help my sister backup her iPhone6, iOS 12.4 and I saw similar error... my question is: should I install the patch like [this how-to ] to avoid this issue? I remember patched this libimobiledevice sometime last Oct. Should I repatch it? Thanks.