libimobiledevice/libimobiledevice #852: Can't "make" on Puppy Linux

00p513-dev, 2019-09-29:
I have installed all dependencies. When i goto `make` it gives me the following error:
```
root# make
make  all-recursive
make[1]: Entering directory '/root/libimobiledevice'
Making all in common
make[2]: Entering directory '/root/libimobiledevice/common'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/libimobiledevice/common'
Making all in src
make[2]: Entering directory '/root/libimobiledevice/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/libimobiledevice/src'
Making all in include
make[2]: Entering directory '/root/libimobiledevice/include'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/libimobiledevice/include'
Making all in tools
make[2]: Entering directory '/root/libimobiledevice/tools'
  CCLD     idevice_id
../src/.libs/libimobiledevice.so: undefined reference to `usbmuxd_save_pair_record_with_device_id'
collect2: error: ld returned 1 exit status
Makefile:652: recipe for target 'idevice_id' failed
make[2]: *** [idevice_id] Error 1
make[2]: Leaving directory '/root/libimobiledevice/tools'
Makefile:454: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/root/libimobiledevice'
Makefile:384: recipe for target 'all' failed
make: *** [all] Error 2
root# 

```

mexmer, 2019-09-30:
seems you are missing libusmuxd or reference to it.

00p513-dev, 2019-11-05:
closing, just installed ubuntu lol

closed 2019-11-05 by 00p513-dev