libimobiledevice/libimobiledevice #855: How to get the Touch ID serial number of device

Johnny314, 2019-10-01:
i am trying to fetch the serial number of Touch ID sensor...
is there any service or mobilegestalt key from which i can get?
@nikias  @FunkyM

mexmer, 2019-10-01:
you can try to find yourself
https://pastebin.com/HY122aPR

Johnny314, 2019-10-03:
i am getting touch id serial number in this format
Cdd2FYJBkA/0D2OgMEfKUg==
its seems like base64 but its not 
however 3utools shows
09D776158241900FF40F63A03047CA52
any ideas how to decode that?

mexmer, 2019-10-04:
if you do base64 decode on your data, you get binary value, write each byte as HEX

Johnny314, 2019-10-04:
ok got it and what is this method work for SysCfgData as well?

here is SysCfgData Dump

Z2ZDU/AEAAAAAAIAAgACAAAAAAA+AAAAI0JMTUYzWTgwNTIwNFlKSFA1UUFjYU1X5Cs0YAwEAAAAAAAAAAAAAGNhTULkKzRgDAUAAAAAAAAAAAAAQlROQ1RBQkMoAQAA2P4BAP////9CVE5DbENGTpQAAABE/gEA/////0JUTkNMQUNXEAcAADT3AQD/////QlROQ0xBQ0IMDQAAKOoBAP////9tTnJTR0hMV0YzUkdKQ0w4AAAAAGxDb04AAAAAAAAAAAAAAAAAAAAAQlROQ3R0YUIRAAAAFOoBAP////9CVE5DRUZBTGQAAACw6QEA/////0JUTkNwTVBM7AEAAMTnAQD/////QlROQ2xhQ1RQAAAAdOcBAP////9CVE5DbENTUFACAAAk5QEA/////0JUTkNsQ3RNbAgAALjcAQD/////QlROQ2xDUEQoAAAAkNwBAP////9CVE5DTENyUAABAACQ2wEA/////3BTR0MBAAAAAAAAAAAAAAAAAAAAI2RvTU5RQUYyAAAAAAAAAAAAAABuZ2VSWkQvQQAAAAAAAAAAAAAAAEdDTEMBAAAAAAAAAAAAAAABAAAAU0hMQwEAAAAAAAAAAAAAAAEAAABxU1JQAQAABAAAAAAAAAAAAAAAAEJUTkNOU3RNLAAAAGTbAQD/////QlROQ0NsZE1bAAAAcNoBAP////9MQ3hSAAIoMgAAAAAAAAAAAAAApEJUTkNTTUNGEQAAAFDbAQD/////QlROQ0JNQ0ZAAAAAENsBAP////9CVE5DU01DQhEAAAD82gEA/////0JUTkNCTUNCMAAAAMzaAQD/////Q0NWQwAAAAC8////KAEAAPAAAABDQ0RDAQAAAFb///+A////r/f//0JUTkNPQVRMFAAAAFzaAQD/////QlROQ2xDU0EkAAAAONoBAP////9CVE5DbENTRyQAAAAU2gEA/////0JUTkNpQ1NBJAAAAPDZAQD/////QlROQ2lDU0ckAAAAzNkBAP////90b1JBAQAAAP8AAAAA/wAAAAD/AHRvUkcBAAAAAQAAAAABAAAAAAEAbENJQQMAAACPEwAAyAAAAIQTAABsQ0lHAgAAAI8TAADIAAAAAAAAAEJUTkNsQ1RHPAAAAJDZAQD/////QlROQ09QUFM4AAAAWNkBAP////9CVE5DTkZTTKQBAAC01wEA/////0JUTkNsQ3BTPAAAAHjXAQD/////QlROQ0NUUkwcAwAAXNQBAP////9CVE5DcFNTTKQBAAC40gEA/////0JUTkN4UlNMpAEAABTRAQD/////QlROQ01QU0ykAQAAcM8BAP////9CVE5DQ1NTTKQBAADMzQEA/////0JUTkNUVFlHcgAAAFjNAQD/////QlROQ1RUUlDeAAAAeMwBAP////9CVE5DbEN4UJAAAADoywEA/////0JUTkNSQUNCMgAAALTLAQD/////QlJUU4EA50r5B3EbRBsAAAAAAABCVE5DYUdUQSQAAACQywEA/////3RvUkMCAAAAAAD/AAEAAAAA/wAAQlROQ01DU0MoAAAAaMsBAP////9CVE5DTENGTKiOAADAPAEA/////2hCd1MFBAAAAAAAAAAAAAAAAAAAY2FNReQrNGAMBgAAAAAAAAAAAAAjZE1SQTE5MDEAAAAAAAAAAAAAAP///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////// [... cut ...] /////wAFAAYAApsA//8ABQAGAASbAP//AAUABgAGTwH//wAJAAQAAAD/AAgABgAAXAH///ADAAQAAAH/8AIACHeMvGWfd44R8AQAAgAJBAAAAEQAAACIEwAAKCMAAOAuAACYOgAAAABkAI2OZQAAAOoB6w7lAQAA9AEqvPQBAADeA2RjxQMAAOgDPrLrAwAA3AWb+8gFAADQB8DhogcAAGAJpHEeCQQAAAAAAGQAN0tlAAAA6gEuo+QBAAD0Achd9QEAAN4DFgDCAwAA6AO99egDAAB8BkPqXQYAANAHPiWgBwAA0Ac+JaAHBAAAAAAAZADRh2UAAADqAUCI5AEAAPQBTev0AQAA3gNwIcIDAADoAyIX6QMAAOIEfczXBAAA4gR9zNcEAADiBH3M1wQEAAAAAABkAPVRZQAAAOoBH77kAQAA9AEJ8vQBAADeAwicwQMAAOgD/rLoAwAA6AP+sugDAADoA/6y6AMAAOgD/rLoAwQAAAA=

mexmer, 2019-10-07:
it's binary data again, but i have no idea, what is inside.

deathlauncher, 2019-10-26:
@Johnny314  Are you able to share how did you get the touch-id serial? mobilegestalt touch-id only gives like true/false

saurav-997, 2021-12-15:
Didn't get touch-id serial, **idevicediagnostics mobilegestalt touch-id** returns true/Success only.

Below is the output i get:- 
<key>MobileGestalt</key>
	<dict>
		<key>Status</key>
		<string>Success</string>
		<key>touch-id</key>
		<true/>
	</dict>

bugstacks, 2022-01-14:
I have the same problem, how can you solve it please.

yunhailu, 2022-01-15:
> Didn't get touch-id serial, **idevicediagnostics mobilegestalt touch-id** returns true/Success only.
> 
> Below is the output i get:-
> 
> ```
> <key>MobileGestalt</key>
> 	<dict>
> 		<key>Status</key>
> 		<string>Success</string>
> 		<key>touch-id</key>
> 		<true/>
> 	</dict>
> ```

Has the problem been solved？