libimobiledevice/libimobiledevice #856: Integrating with OSS-Fuzz

Google-Autofuzz, 2019-10-02:
Greetings libimobiledevice developers and contributors,

We’re reaching out because your project is an important part of the open source ecosystem, and we’d like to invite you to integrate with our [fuzzing](https://www.owasp.org/index.php/Fuzzing) service, [OSS-Fuzz]( https://opensource.googleblog.com/2016/12/announcing-oss-fuzz-continuous-fuzzing.html ). OSS-Fuzz is a free fuzzing infrastructure you can use to identify security vulnerabilities and stability bugs in your project. OSS-Fuzz will:

- Continuously run at scale all the fuzzers you write.
- Alert you when it finds issues.
- Automatically close issues after they’ve been fixed by a commit.

Many widely used [open source projects]( https://github.com/google/oss-fuzz/tree/master/projects ) like OpenSSL, FFmpeg, LibreOffice, and ImageMagick are fuzzing via OSS-Fuzz, which helps them find and remediate [critical issues]( https://bugs.chromium.org/p/oss-fuzz/issues/list?can=1&q=status%3AFixed%2CVerified+Type%3ABug%2CBug-Security+-component%3AInfra+ ). 

Even though typical integrations can be done in < 100 LoC, we have a [reward program]( https://www.google.com/about/appsecurity/patch-rewards/ ) in place which aims to recognize folks who are not just contributing to open source, but are also working hard to make it more secure.

We want to stress that anyone who meets the eligibility criteria and integrates a project with OSS-Fuzz is eligible for a reward.

If you're not interested in integrating with OSS-Fuzz, it would be helpful for us to understand why—lack of interest, lack of time, or something else—so we can better support projects like yours in the future.

If we’ve missed your question in our [FAQ]( https://google.github.io/oss-fuzz/faq/ ), feel free to reply or reach out to us at oss-fuzz-outreach@googlegroups.com.

Thanks!

Tommy
OSS-Fuzz Team

Google-Autofuzz, 2019-11-13:
We can also help by starting the OSS-Fuzz integration process for you if lack of time. We just need a google email from you to send bugs to you when libfuzzer finds any.

nikias, 2019-11-13:
I would love to add support for it. I already integrated oss-fuzz into libplist and it found quite a lot of bugs. I will see how I can add it. You should have my gmail from libplist.

Google-Autofuzz, 2019-11-18:
````
checking for libusbmuxd... no
configure: error: Package requirements (libusbmuxd >= 1.1.0) were not met:

No package 'libusbmuxd' found

````
When I compile the package, I got the error as shown above. Why is the package libusbmuxd not found? I basically compiled by doing the following:

````
./autogen.sh --prefix=/opt/local --enable-debug-code
make
sudo make install
````