libimobiledevice/libimobiledevice #857: Unhandled Status Code 133 on iPhone 11 (N104AP)

iM4CH3T3, 2019-10-03:
Getting Unhandled Status Code 133 on iPhone 11 (N104AP) when trying to restore iOS 13.1.2 
[image]

All your tools on latest version , just installed everything today freshly

nikias, 2019-10-03:
Can you run with -d ? Then I can see the whole request to the TSS.

iM4CH3T3, 2019-10-03:
That’s with -d 

[image]

nikias, 2019-10-03:
Can you please paste the plist output as text and not as an image? You might want to use a gist or attach it as a file since it's probably large.

iM4CH3T3, 2019-10-04:
I hope u can open it 

https://t.co/J0OGkbZZXk?amp=1

nikias, 2019-10-04:
You are definitely not using the latest source of idevicerestore. I could tell from `libauthinstall-293.1.16` and the missing tags for `@eUICC,Ticket` etc.

nikias, 2019-10-04:
Also you posted this ticket in the wrong project, this usually needs to go to idevicerestore :)

iM4CH3T3, 2019-10-04:
So new version should fix that right ?
Sorry btw

nikias, 2019-10-04:
That's fine but posting issues when not running the latest version isn't really helpful, since I was going to search for a problem that is probably not there anymore. If this problem persists even with latest code, open a new ticket on https://github.com/libimobiledevice/idevicerestore/issues.

closed 2019-10-04 by nikias

iM4CH3T3, 2019-10-04:
Thx very much 
Sorry again for wasting your time 🙈