libimobiledevice/libimobiledevice #861: idevicesyslog not showing logs on iOS13

christopherfujino, 2019-10-10:
Using `idevicesyslog` I am not able to see any application logs from an iPhone Xr with iOS 13.1. The only logs I can see are when I first connect the phone via USB [here](https://gist.github.com/christopherfujino/6e22a26d15c91fadd9feaf8fde97c5b4).

I can see app logs when I run the app via Xcode. This is causing https://github.com/flutter/flutter/issues/41133.

nikias, 2019-10-10:
If you run the app via Xcode, you get direct stdout and stderr output via debugserver. The output via idevicesyslog is whatever the device sends and cannot be controlled from the client side. Apple decided to silence app log output on iOS 9 or 10 iirc.

christopherfujino, 2019-10-10:
Hmm, I'm not sure what you mean by "Apple decided to silence app log output on iOS 9 or 10." FWIW I am able to read app logs in `idevicesyslog` from the same app run on iOS 12.4.2.

nikias, 2019-10-10:
Well then that must be another change in iOS 13, but it doesn't change the fact that the syslog service ("syslog relay") just relays whatever the device has for you. If the app output isn't there idevicesyslog can't change that. Have you checked the console.app output for your device or via Xcode -> Window -> Devices and Simulators -> select device -> system log?

christopherfujino, 2019-10-11:
Ahh, so I'm not seeing my app logs under console.app either. We will try to address this by switching to using OS log. Thank you for help @nikias!

closed 2019-10-11 by christopherfujino