libimobiledevice/libimobiledevice #862: idevicediagnostics KEY

HuangTao15521192185, 2019-10-12:
As we know,when we use the command, idevicediagnostics ioregentry [KEY],it can print IORegistry entry of device.The KEYs include AppleARMPMUCharger, ASPStorage and so on.Now I need to know more about [KEY] in order that I can get more information about screen or LCD for iPads,because I couldn't get anything useful by [KEY] that is "product" or "lcd".
Please help.

mroger-tech, 2019-12-30:
i need too.

tangwanchao, 2020-07-29:
i need too.

saurav-997, 2021-04-15:
i need too.

msk521, 2021-12-10:
i need too

msk521, 2021-12-10:
i need too.

uncommonthinker, 2022-10-22:
# NOTE that all keys and values shown are for MacOSX and because of the architecture similarity few values are the same.
I think I have found a way to look for keys, you would need a Mac system to do that. I have the latest version of libimobiledevice which I have installed through brew on my Apple M1 2020, OSX Monteray.

1. Firstly download, https://github.com/vulgo/IORegistryExplorer
2. This tool actually shows keys for MacOSX, but as the base architecture is the same you would be able to see a lot of keys that are similar to that of iOS.
3. Find some of the keys that I found while going through the list.

IORegKeys
****
AppleAPFSContainer
AppleStockholmControl
AppleSPUAppDriver
AppleSPUProfileDriver
AppleSPU
AppleSPUHIDDevice
AppleSPUFirmwareService
****
# Audio
AppleAOPAudioController
AppleAOPVoiceTriggerController
****
# Interfaces
IOPP
IOHIDInterface
AppleARMPE
****
AVD
AppleAVD
AppleAVE2Driver
****
IODARTMapper
RTBuddyService
****
IOAccessoryTRM
IOAccessoryUSBConnectShim
USB2
USB3
AppleJPEGDriver
AGXArmFirmwareMapper
****
AppleT8015TempSensor
AppleMobsiTmpSADC
****
AppleDialogSPMIPMU
AppleDialogSPMIPMURTC
****
# CLOSED LOOP PERFORMANCE CONTROLLER
AppleCLPC
****
AppleMesaSEPDriver
AppleBiometricServices
AppleSEPXARTService
****
AppleSMC
AppleARMPMUPowerSensor
AppleARMPMUTempSensor
AppleSmartBatteryManager
AppleSMCPMU
****
AppleDiagnosticDataAccessReadOnly
AppleMesaShim
AppleMultitouchDevice
****
IOSerialBSDClient
AppleSerialShim
****
IONetworkStack
AppleBCMWLANBusInterfacePCIe
AppleBCMWLANCore
****
AppleOLYHAL
AppleARMBacklight
AppleM68Buttons
AppleARMCPU
AppleMobileApNonce
AppleImage4
****
product
sacm
AppleARMSlowAdaptiveClockingManager
IOResources

Hope this helps everyone looking for a way around the keys!