libimobiledevice/libimobiledevice #864: I use mobileSync service to add contact to my iphone,but get "DLMessageDisconnect" message

Roger-mazheng, 2019-10-22:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "[图片]http://www.apple.com/DTDs/PropertyList-1.0.dtd
">
<plist version="1.0">
<array>
        <string>SDMessageProcessChanges</string>
        <string>com.apple.Contacts</string>
        <dict>
                <key>0</key>
                <dict>
                        <key>com.apple.syncservices.RecordEntityName</key>
                        <string>com.apple.contacts.Contact</string>
                        <key>birthday</key>
                        <date>2001-01-01T00:00:00Z</date>
                        <key>company name</key>
                        <string>company</string>
                        <key>display as company</key>
                        <string>person</string>
                        <key>first name</key>
                        <string>ma</string>
                        <key>last name</key>
                        <string>zheng</string>
                        <key>notes</key>
                        <string>hello</string>
                        <key>job title</key>
                        <string>aaaaaa</string>
                        <key>department</key>
                        <string>dep</string>
                </dict>
        </dict>
        <false/>
        <dict>
                <key>SyncDeviceLinkAllRecordsOfPulledEntityTypeSentKey</key>
                <true/>
                <key>SyncDeviceLinkEntityNamesKey</key>
                <array>
                        <string>com.apple.contacts.Contact</string>
                        <string>com.apple.contacts.Group</string>
                </array>
        </dict>
</array>
</plist>
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\prop
erty_list_service.c:200 internal_plist_receive_timeout(): initial read=4
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\prop
erty_list_service.c:206 internal_plist_receive_timeout(): 98 bytes following
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\prop
erty_list_service.c:219 internal_plist_receive_timeout(): received 98 bytes
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\prop
erty_list_service.c:245 internal_plist_receive_timeout(): printing 274 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "[图片]http://www.apple.com/DTDs/PropertyList-1.0.dtd
">
<plist version="1.0">
<array>
        <string>DLMessageDisconnect</string>
        <string>Connection is being deleted</string>
</array>
</plist>
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\prop
erty_list_service.c:132 internal_plist_send(): sending 104 bytes
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\serv
ice.c:144 service_send(): sending 4 bytes
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\serv
ice.c:144 service_send(): sending 104 bytes
22:38:56 F:\newstart\libimobiledevice-vs-master\libimobiledevice-vs-master\libimobiledevice\src\prop
erty_list_service.c:147 internal_plist_send(): ERROR: sending to device failed.bytes=[0],sizeof(nlen
)=[4]

what is "Connection is being deleted" mean? and what can i do?

mexmer, 2019-10-22:
did you enable contact sync with desktop?

Roger-mazheng, 2019-10-23:
@mexmer No,what does "...with desktop" mean?  i can get contacts and delete contacts,but only can not add contacts.

mroger-tech, 2019-10-23:
> did you enable contact sync with desktop?

thanks for your reply anyway,can you tell me the detail about add contacts to iphone?

thank you very much!!!

mexmer, 2019-10-23:
you need to check value of com.apple.mobile.tethered_sync
if it contains DisableTethered - false or 0 (value depends on version of ios) under "Contacts", you can sync, if it's True or 1, then you can't sync with PC and need to change it (if it's 1, then set 0, if it's True, set False).

after that, sync should work.

synced contacts should appear in "From my computer" category, or something like that.

mroger-tech, 2019-10-24:
> you need to check value of com.apple.mobile.tethered_sync
> if it contains DisableTethered - false or 0 (value depends on version of ios) under "Contacts", you can sync, if it's True or 1, then you [...]

Thanks!

I will try your method.

mroger-tech, 2019-10-24:
> Thanks!
>
> I will try your method.

[image]

I use "ideviceinfo.exe -x -q com.apple.mobile.tethered_sync" to check the value, but get nothing(iphone5s,ios12.4.2)

mexmer, 2019-10-24:
that means, it's on default values, which is disabled sync with desktop

should look like this
[image]

mroger-tech, 2019-11-20:
> that means, it's on default values, which is disabled sync with desktop
> 
> should look like this
> [image]

I'm sorry to bother you again,I try to set value of com.apple.mobile.tethered_sync like yours, but the error still exist,when i call "mobilesync_remap_identifiers" ,the error occur,and the return error code is -2.

[image]

mexmer, 2019-11-21:
detailed log will help
-2 is MOBILESYNC_E_PLIST_ERROR, which means response from device is invalid, but it might also mean, there is no remap identifier map in response. for example if you don't change any data, it will not return new map.

mroger-tech, 2019-11-22:
hello mexmer,detail log bellow:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

mexmer, 2019-11-22:
from your log seems you sent only one contact, then disconeccted. you can't call remap after disconnect, remap is always called after sending changes, so it will tell you record was remapped to which id.

although issue might be also, that you using ID 0 for new record, you must specify positive integer value, that is higher than last id of received record. since you get no data, you will use 1, not 0, that's probly why instead of confirming new record, connection was closed.

when i add new entries to phone, i use lastentityindex + 100000, reason is, that cloud synchronized contacts cannot be retrieved with mobilesync, and i don't want to risk overwriting cloud contacts.

mroger-tech, 2019-11-25:
> from your log seems you sent only one contact, then disconeccted. you can't call remap after disconnect, remap is always called after sending changes, so it will tell you record was remapped to [...]

1. In fact, after i sent changes (call mobilesync_send_changes),i call the function mobilesync_remap_identifiers,and the function(mobilesync_remap_identifiers)give me "disconnect"response back.

2.I modify my code, i use  ID 100000  for new record, but still get "disconnect" response back.

Note: I removed all record from my iphone.

mexmer, 2019-11-28:
when i sync contact i do following, first i query state of sync with computer from lockdown

query
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Domain</key>
<string>com.apple.mobile.tethered_sync</string>
<key>Label</key>
<string>Apple Client version 1.0</string>
<key>Request</key>
<string>GetValue</string>
</dict>
</plist>
```
response
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>Contacts</key>
<dict>
<key>DisableTethered</key>
<false/>
<key>SyncingOS</key>
<string>Windows</string>
</dict>
</dict>
</plist>
```
if disablethered is true or not present, you need to set it to false.

then i do following steps

1) start com.apple.mobilesync service, communication over mobilesync link (which needs to be started with SSL as of IOS 13.0) looks as following (mind communication log is from software based on libimobile, so looks somewhat different) and perform version exchange - equivalent to  **mobilesync_client_start_service()**

2) request data sync from device - equivalent to **mobilesync_start()**

3) application receives all contacts from device  - equivalent to **mobilesync_get_all_records_from_device()** after you need to send **mobilesync_acknowledge_changes_from_device()**

4) change direction of changes - equivalent to **mobilesync_ready_to_send_changes_from_computer()**

5) send all changed record to device - equivalent to **mobilesync_send_changes()**

6) receive map of remapped record ids - equivalent to **mobilesync_remap_identifiers()**

7) finish communication - equivalent to **mobilesync_finish()**

mroger-tech, 2019-12-15:
> when i sync contact i do following, first i query state of sync with computer from lockdown
> 
> query
> 
> ```
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist version="1.0">
> <dict>
> <key>Domain</key>
> <string>com.apple.mobile.tethered_sync</string>
> <key>Label</key>
> <string>Apple Client version 1.0</string>
> <key>Request</key>
> <string>GetValue</string>
> </dict>
> </plist>
> ```
> 
> response
> 
> ```
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist version="1.0">
> <dict>
> <key>Contacts</key>
> <dict>
> <key>DisableTethered</key>
> <false/>
> <key>SyncingOS</key>
> <string>Windows</string>
> </dict>
> </dict>
> </plist>
> ```
> 
> if disablethered is true or not present, you need to set it to false.
> 
> then i do following steps
> 
> 1. start com.apple.mobilesync service, communication over mobilesync link (which needs to be started with SSL as of IOS 13.0) looks as following (mind communication log is from software based on libimobile, so looks somewhat different) and perform version exchange - equivalent to  **mobilesync_client_start_service()**
> 2. request data sync from device - equivalent to **mobilesync_start()**
> 3. application receives all contacts from device  - equivalent to **mobilesync_get_all_records_from_device()** after you need to send **mobilesync_acknowledge_changes_from_device()**
> 4. change direction of changes - equivalent to **mobilesync_ready_to_send_changes_from_computer()**
> 5. send all changed record to device - equivalent to **mobilesync_send_changes()**
> 6. receive map of remapped record ids - equivalent to **mobilesync_remap_identifiers()**
> 7. finish communication - equivalent to **mobilesync_finish()**

As the log i show you before ,when i add contact to iphone , I did the same steps as yours. but i  never get the right response.I have no idea any more, can you show me your detail log when you add contacts?

Thank you very much!

mexmer, 2019-12-18:
```xml
73.1141 [6:data:10292] Received plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageVersionExchange</string>
<integer>300</integer>
<integer>100</integer>
</array>
</plist>
73.1141 [6:data:10292] Sending XML plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageVersionExchange</string>
<string>DLVersionsOk</string>
<integer>300</integer>
<integer>100</integer>
</array>
</plist>
73.1175 [6:data:10292] Received plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageDeviceReady</string>
</array>
</plist>
73.1176 [6:data:10292] Sending XML plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>SDMessageSyncDataClassWithDevice</string>
<string>com.apple.Contacts</string>
<string>---</string>
<string>2019-12-18 12-55-34 +0100</string>
<integer>106</integer>
<string>___EmptyParameterString___</string>
</array>
</plist>
73.1584 [6:data:10292] Received plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>SDMessageSyncDataClassWithComputer</string>
<string>com.apple.Contacts</string>
<string>2019-12-18 12-54-55 +0100</string>
<string>Dec 18 2019 12:55:34 +0100</string>
<string>SDSyncTypeSlow</string>
<integer>106</integer>
<string>___EmptyParameterString___</string>
</array>
</plist>
73.1584 [5:link:10292] 13D9AA10 Link timeout for link (1060/49602) set to 10000 (0)
73.1585 [6:data:10292] Sending XML plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>SDMessageGetChangesFromDevice</string>
<string>com.apple.Contacts</string>
</array>
</plist>
 73.1627 [6:data:10292] Received plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
[... 137 lines cut ...]
<plist version="1.0">
<array>
<string>SDMessageFinishSessionOnDevice</string>
<string>com.apple.Contacts</string>
</array>
</plist>
 73.3084 [6:data:10292] Received plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>SDMessageDeviceFinishedSession</string>
<string>com.apple.Contacts</string>
</array>
</plist>
73.8086 [6:data:10292] Sending XML plist: 
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
<string>DLMessageDisconnect</string>
<string>All done, thanks for the memories</string>
</array>
</plist>
```

mroger-tech, 2019-12-19:
> ```
> 73.1141 [6:data:10292] Received plist:
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist [...]

Thanks for you reply, I will carefully compare your logs to find out what's wrong with me.

mroger-tech, 2019-12-23:
I finally found the problem. It was the encoding of my text. I changed the encoding to utf8 and got it right.
It was a stupid mistake.
Finally, thanks for @mexmer's continued help.

closed 2019-12-23 by mroger-tech

mexmer, 2019-12-31:
good to know there is no issue with library itself. and yes, when you use text (plist) format you must use UTF-8

zuodd, 2024-09-09:
> good to know there is no issue with library itself. and yes, when you use text (plist) format you must use UTF-8

Hi @mexmer Is there any way to get contacts via imobiledevice binary library without coding?

zuodd, 2024-09-09:
> good to know there is no issue with library itself. and yes, when you use text (plist) format you must use UTF-8

I mean just use cmd from the terminal but no resource file compile and coding.