libimobiledevice/libimobiledevice #868: Unable to compile for Windows

anshumanchatterji, 2019-10-30:
I tried compiling libimobiledevice & its dependancies using both cygwin and MinGW. After solving all the issues, I was able to create the windows executables. However, for both the environments(Cygwin & MinGW) I encontered the same error:

```
./idevice_id -l
socket_connect: getaddrinfo: Unknown error
ERROR: Unable to retrieve device list!
```
Same error was also reported here https://github.com/libimobiledevice/libimobiledevice/issues/783

Is there some writeup for compiling in windows that i can follow? Below is the compilation of various steps that i followed:

> 
> Install MSys2
>   https://stackoverflow.com/questions/26453293/what-is-the-best-way-to-install-pkg-config-on-mingw-in-2014
>   https://www.msys2.org/
>   
> Install MinGW
>   http://www.mingw.org/wiki/MSYS
>   Set bin folder to environemnt variables in windows
>   
>   
> Build libplist
>   ./autogen.sh
>   make
>   make install
> (I remember getting python errors, so had to install --without-cython)
>   
> Build usbmuxd
>   ./autogen.sh
>   make
>   make install
>   
> Install GNUTLS as unable to find openssl dev libs
>   https://github.com/libimobiledevice/libimobiledevice/issues/617
>   pacman -S mingw-w64-x86_64-gnutls
>   pacman -S gnutls
>   pacman -S libgnutls
>   pacman -S openssl
>   pacman -S libopenssl
>   pacman -S openssl-devel
> 
> Install libcrypt
>  pacman -S libgcrypt-devel
> 
>      
> Build
>    ./autogen.sh --without-cython --disable-openssl
>   make
>   make install

mexmer, 2019-10-30:
use msys2, don't use cygwin or mingw. i mentioned that in linked issue.

anshumanchatterji, 2019-10-30:
@mexmer I had indeed used the MSys2 terminal. Do i need to remove mingw and retry building on Msys2 terminal from scratch?

anshumanchatterji, 2019-10-31:
I was able to compile with MSys2 with OpenSSL.

This might help someone else:

**Install MSys2**
  https://stackoverflow.com/questions/26453293/what-is-the-best-way-to-install-pkg-config-on-mingw-in-2014
  https://www.msys2.org/
  
  
**Build libplist**
  git clone https://github.com/libimobiledevice/libplist.git
  (I remember getting python errors, so had to install --without-cython)
  ./autogen.sh --without-cython
  make
  make install

**Build libusb**
  git clone https://github.com/libusb/libusb.git
  ./autogen
  make
  make install
  
  
  
**Build libusbmuxd**
  git clone https://github.com/libimobiledevice/libusbmuxd.git
  ./autogen.sh
  make
  make install

**Install libcrypt**
 pacman -S libgcrypt-devel

**With OpenSSL support which is by default:**
  pacman -S openssl
  pacman -S libopenssl
  pacman -S openssl-devel
  

**Install GNUTLS if unable to find openssl dev libs**
  https://github.com/libimobiledevice/libimobiledevice/issues/617
  pacman -S mingw-w64-x86_64-gnutls
  pacman -S gnutls
  pacman -S libgnutls
  pacman -S libgnutls-devel
  

     
**Build libimobiledevice**
  git clone https://github.com/libimobiledevice/libimobiledevice.git
  _// With OpenSSL_
    ./autogen.sh --without-cython
  _// With GNU TLS_
   ./autogen.sh --without-cython --disable-openssl
  make
  make install

closed 2019-10-31 by anshumanchatterji

igor84, 2020-03-16:
This is the list of things I had to install in order to build everything above without python issues:

pacman -S git
pacman -S libedit
pacman -S mingw-w64-i686-toolchain
pacman -S --needed base-devel msys2-devel
pacman -S mingw-w64-x86_64-python
pacman -S msys/libcrypt-devel
pacman -S openssl
pacman -S libopenssl
pacman -S openssl-devel

In order to also build https://github.com/libimobiledevice/ideviceinstaller I had to do the following as well:

pacman -S mingw-w64-x86_64-libzip
cp /mingw64/lib/pkgconfig/libzip.pc /usr/lib/pkgconfig/
// Delete the line containing invalid D:/msys64 path from copied libzip.pc
sed -i '/-lD:\/msys64/d' /usr/lib/pkgconfig/libzip.pc
git clone https://github.com/libimobiledevice/ideviceinstaller.git
cd ideviceinstaller
// Remove switch to threat warnings as errors since there are some warnings
sed -i 's/ -Werror//g' configure.ac
./autogen.sh
make
makeinstall

winstondu, 2020-08-17:
@anshumanchatterji , once you build the project, how do you use it as a library when you develop your own program?

anshumanchatterji, 2021-02-04:
@winstondu I took the exes and have used it in my dotnet project by consuming the output the processes invoked as command.

andrei219, 2021-03-26:
> @anshumanchatterji , once you build the project, how do you use it as a library when you develop your own program?

Did you build it ? How you did it ? I can't =(