libimobiledevice/libimobiledevice #870: Problem with IOS 13.2.2

kursatyurt, 2019-11-10:
Everytime when I try connect laptop it ask for trusting 

File transfer is impossible it stucks even if I see what is inside my tablet it does not copy-paste anything 

> [kursatyurt@kursatyurt ~]$ idevicepair pair
ERROR: Please accept the trust dialog on the screen of device 00008020000D05202EBB002E, then attempt to pair again.
[kursatyurt@kursatyurt ~]$ idevicepair validate
ERROR: Device 00008020000D05202EBB002E returned unhandled error code -8

nikias, 2019-11-10:
What platform are you on? The udid is incorrect, it needs a - in it and usually you don't have to call idevicepair at all, usbmuxd is handling it (on both macOS and linux)

kursatyurt, 2019-11-10:
On Fedora 30 Linux. Up to today I just connect and transfer it works well but now it does not transfer anything. I actually see everything ? But it does not copy 
[](https://pasteboard.co/IG5plMW.png)

[](https://pasteboard.co/IG5pyKh.png)

mexmer, 2019-11-11:
this looks more like issue with tool you are using, maybe some update did break it.

kursatyurt, 2019-11-11:
Might be I am looking for solution. I remove all libimobiledevice stuff and reinstall them but it does not work

kursatyurt, 2019-11-13:
I have downgrade 13.2 but it does not work. I also checked with my friends Iphone it works. So, how can be solve this problem ? Any Suggestion

fredriccliver, 2019-11-18:
I also got a problem.

`ideviceinfo returned an error:
ERROR: Could not connect to lockdownd, error code -8`

idevicepair pair 
ERROR: Could not connect to lockdownd, error code -8

and furthermore, My iPhone had a stucked, and some app doesn't work so I had had to force reboot my iPhone

closed 2022-08-31 by kursatyurt