libimobiledevice/libimobiledevice #871: Build issue on Raspberry Pi

JoshGuyette, 2019-11-12:
make[2]: Entering directory '/home/pi/iPhone/src/tmp/libimobiledevice/src'
  CC       idevice.lo
idevice.c: In function ‘usbmux_event_cb’:
idevice.c:194:19: error: ‘usbmuxd_device_info_t’ {aka ‘const struct <anonymous>’} has no member named ‘conn_type’
  if (event->device.conn_type == CONNECTION_TYPE_USB) {
                   ^
idevice.c:194:33: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
  if (event->device.conn_type == CONNECTION_TYPE_USB) {
                                 ^~~~~~~~~~~~~~~~~~~
                                 CONNECTION_USBMUXD
idevice.c:194:33: note: each undeclared identifier is reported only once for each function it appears in
idevice.c:196:26: error: ‘usbmuxd_device_info_t’ {aka ‘const struct <anonymous>’} has no member named ‘conn_type’
  } else if (event->device.conn_type == CONNECTION_TYPE_NETWORK) {
                          ^
idevice.c:196:40: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function); did you mean ‘CONNECTION_NETWORK’?
  } else if (event->device.conn_type == CONNECTION_TYPE_NETWORK) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
                                        CONNECTION_NETWORK
idevice.c: In function ‘idevice_get_device_list_extended’:
idevice.c:249:18: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_type’
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                  ^
idevice.c:249:32: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                                ^~~~~~~~~~~~~~~~~~~
                                CONNECTION_USBMUXD
idevice.c:252:25: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_type’
   } else if (dev_list[i].conn_type == CONNECTION_TYPE_NETWORK) {
                         ^
idevice.c:252:39: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function); did you mean ‘CONNECTION_NETWORK’?
   } else if (dev_list[i].conn_type == CONNECTION_TYPE_NETWORK) {/
                                       ^~~~~~~~~~~~~~~~~~~~~~~
                                       CONNECTION_NETWORK
idevice.c:254:32: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_data’
    size_t addrlen = dev_list[i].conn_data[0];
                                ^
idevice.c:256:52: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_data’
    memcpy(newlist[newcount]->conn_data, dev_list[i].conn_data, addrlen);
                                                    ^
idevice.c: In function ‘idevice_get_device_list’:
idevice.c:302:18: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_type’
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                  ^
idevice.c:302:32: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                                ^~~~~~~~~~~~~~~~~~~
                                CONNECTION_USBMUXD
idevice.c: In function ‘idevice_from_mux_device’:
idevice.c:348:16: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_type’
  switch (muxdev->conn_type) {
                ^~
idevice.c:349:7: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
  case CONNECTION_TYPE_USB:
       ^~~~~~~~~~~~~~~~~~~
       CONNECTION_USBMUXD
idevice.c:353:7: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function); did you mean ‘CONNECTION_NETWORK’?
  case CONNECTION_TYPE_NETWORK:
       ^~~~~~~~~~~~~~~~~~~~~~~
       CONNECTION_NETWORK
idevice.c:355:33: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_data’
   size_t len = ((uint8_t*)muxdev->conn_data)[0];
                                 ^~
idevice.c:357:35: error: ‘usbmuxd_device_info_t’ {aka ‘struct <anonymous>’} has no member named ‘conn_data’
   memcpy(device->conn_data, muxdev->conn_data, len);
                                   ^~
idevice.c:348:2: warning: switch missing default case [-Wswitch-default]
  switch (muxdev->conn_type) {
  ^~~~~~
idevice.c: In function ‘idevice_new_with_options’:
idevice.c:372:21: error: ‘DEVICE_LOOKUP_USBMUX’ undeclared (first use in this function); did you mean ‘IDEVICE_LOOKUP_USBMUX’?
   usbmux_options |= DEVICE_LOOKUP_USBMUX;
                     ^~~~~~~~~~~~~~~~~~~~
                     IDEVICE_LOOKUP_USBMUX
idevice.c:375:21: error: ‘DEVICE_LOOKUP_NETWORK’ undeclared (first use in this function); did you mean ‘IDEVICE_LOOKUP_NETWORK’?
   usbmux_options |= DEVICE_LOOKUP_NETWORK;
                     ^~~~~~~~~~~~~~~~~~~~~
                     IDEVICE_LOOKUP_NETWORK
idevice.c:378:21: error: ‘DEVICE_LOOKUP_PREFER_NETWORK’ undeclared (first use in this function); did you mean ‘IDEVICE_LOOKUP_PREFER_NETWORK’?
   usbmux_options |= DEVICE_LOOKUP_PREFER_NETWORK;
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                     IDEVICE_LOOKUP_PREFER_NETWORK
idevice.c:380:12: warning: implicit declaration of function ‘usbmuxd_get_device’; did you mean ‘usbmuxd_get_device_list’? [-Wimplicit-function-declaration]
  int res = usbmuxd_get_device(udid, &muxdev, usbmux_options);
            ^~~~~~~~~~~~~~~~~~
            usbmuxd_get_device_list
make[2]: *** [Makefile:569: idevice.lo] Error 1

Quick fix for building libimobiledevice on Raspberry Pi

sudo apt-get install libplist++
git revert -n 7dbc17f2c763a7b18bc1a867f39918c26bb5e9e5
git revert -n d6ca8fca11c6aa86ed5fbcf943e1580eb8587538
git revert -n fe3ef93359d9a1105b920d4a7a342706bd3b32c0
git revert -n 6a73100414f66d12287105a322529c488f32182e
git revert -n 74dfc6485b61c76775c332f0ae6c65a28bcced9c
git revert -n 953c13961071a9836430a814159aae33502e2e0c
git revert -n 5390060ec90bed0f565698a389cebc57c09c3ac8
./autogen.sh
make

nikias, 2019-11-13:
Hey, do not revert commits. That is not a fix. Instead, update the dependencies, as libusbmuxd which is the reason it doesn't compile for you.

closed 2019-11-25 by JoshGuyette

zyf0330, 2020-01-02:
I use Ubuntu 19.10, and meet this same problem. I install newest libusbmuxd-dev but not work.
I checkout to 5390060^ and it makes successfully.