libimobiledevice/libimobiledevice #872: idevicedebug error: 02 in iOS 13.2.2

hduarte, 2019-11-12:
By executing idevicedebug run <bundle id> I get:
```
idevicedebug run <bundle id>
ERROR: 02
```

iPhone SE with iOS 13.2.2

zalmoxis, 2019-11-22:
I have the same problem... seems 13.2.2 and 13.2.3 both give "ERROR: 02"

NirBY, 2019-11-26:
Adding here the debug logs, hope it helps:
[idevicedebug.txt](https://github.com/libimobiledevice/libimobiledevice/files/3890859/idevicedebug.txt)

anavarrog, 2019-12-09:
Same problem here.

bikee1235, 2019-12-18:
Getting the same error while using this command

**idevicedebug -u udid run com.bundle.id**

**ERROR: 02**

jpstotz, 2020-01-17:
Using the latest version from master branch and applying the diff of #860 fixed this problem for me.

idevicedebug run now works correctly with two 13.3 devices.

michelleangela, 2020-01-25:
> Adding here the debug logs, hope it helps:
> [idevicedebug.txt](https://github.com/libimobiledevice/libimobiledevice/files/3890859/idevicedebug.txt)

Same problem with iOS 13.3, I get the same call stack too.

1814625556, 2020-05-19:
> Using the latest version from master branch and applying the diff of #860 fixed this problem for me.
> 
> `idevicedebug run` now works correctly with two 13.3 devices.
> 
> Edit: At least sometimes `idevicedebug run` works. In some cases the app tries to start: you see something appearing on the screen for some milliseconds but then this screen vanishes. In the end the app does not really start.
> 
> Edit2: I found out why idevicedebug sometimes does not work as it should and made a PR to fix that: #906. It is totally unrelated to #860 and therefore in my opinion both PRs are necessary for making idevicedebug working again and working reliable.

could you give me a useful package?

michelleangela, 2020-10-09:
Are there any updates on resolving this issue?