libimobiledevice/libimobiledevice #874: idevicecrashreport and idevicescreenshot failed in IOS(13+)

jiaoyujulx, 2019-11-15:
Logs:
16:26:54.313 [main] INFO  util.Util - [idevicecrashreport -u 00008030-0002399A36C0802E -k /Users/mt2/Desktop/bengkui/UICrawler/UICrawler-master/target/crawler_output/00008030-0002399A36C0802E-2019-11-14-16_09_35/ips]  ---output is : Usage: idevicecrashreport [OPTIONS] DIRECTORY
Move crash reports from device to a local DIRECTORY.

  -e, --extract		extract raw crash report into separate '.crash' file
  -k, --keep		copy but do not remove crash reports from device
  -d, --debug		enable communication debugging
  -u, --udid UDID	target specific device by its 40-digit device UDID
  -h, --help		prints usage information

Homepage: <http://libimobiledevice.org>
+++++++++++++++++++usage: idevicecrashreport [options] directory
move crash reports from device to a local directory.

  -e, --extract		extract raw crash report into separate '.crash' file
  -k, --keep		copy but do not remove crash reports from device
  -d, --debug		enable communication debugging
  -u, --udid udid	target specific device by its 40-digit device udid
  -h, --help		prints usage information
idevicescreenshot logs is simliar with it.Is the problem of uuid?I guess uuid in ios13+ is not 40-digit,ios12+ is 40-digit.This is different

mexmer, 2019-11-18:
UDID is 40 letter for all iphones before XR/XS release (2018)
UDID is 40 letter for all ipads released before 2019

UDID in new form 8symbols-16symbols is for all iPHONE models starting from XR/XS (september 2018) and all ipad models starting march 2019 (ipad air 3, ipad pro 2019, ipad 2019)

libimobile and it's tools were already patched for this change, so your problem might be, you not using current version of libimobile, but old one

UDID change has nothing with iOS version, only with device model.

nikias, 2019-11-19:
Hi @jiaoyujulx, as @mexmer pointed out you're not using the latest codebase. The tools help doesn't whos "40-digit device udid" in the help anymore. So update your code. Also remember to update libusbmuxd too.

jiaoyujulx, 2019-11-19:
Thank you.The problem is resolved by reinstalling the lastest version of libmobiledevice.

closed 2019-11-19 by jiaoyujulx