libimobiledevice/libimobiledevice #879: GNUTLS backend not working with GNUTLS 3.6.10

hadess, 2019-11-25:
Support for GNUTLS 3.6.0 was fixed in 99f4a332bbfa6886e300318381cb9efa972ca0ac and 90d4399c93ad73871db02eec115f0ac7f6137213 however ifuse and gvfs-afc don't work with more recent versions of libimobiledevice and newer GNUTLS on my iPhone 6s.

I believe this has to do with the TLS 1.3 support added in 3.6.3:
https://lists.gnupg.org/pipermail/gnutls-devel/2018-July/008584.html
https://nikmav.blogspot.cz/2018/05/gnutls-and-tls-13.html

Equivalents to a9ccd31ddd5d59db8dc5114c7c85e3eac8743578 and a9e69b1252e5918b6d8ada1209ccefde301cfa26 are probably needed.

I've now switched libimobiledevice on Fedora to use openssl, as that's what's actually getting exercised upstream, but other distributions might not have the opportunity.

nikias, 2019-11-25:
I wonder if this would do:
```
diff --git a/src/idevice.c b/src/idevice.c
index 31a8d3a..cd6258d 100644
--- a/src/idevice.c
+++ b/src/idevice.c
@@ -982,7 +982,7 @@ LIBIMOBILEDEVICE_API idevice_error_t idevice_connection_enable_ssl(idevice_conne
        gnutls_certificate_client_set_retrieve_function(ssl_data_loc->certificate, internal_cert_callback);
 #endif
        gnutls_init(&ssl_data_loc->session, GNUTLS_CLIENT);
-       gnutls_priority_set_direct(ssl_data_loc->session, "NONE:+VERS-TLS1.0:+ANON-DH:+RSA:+AES-128-CBC:+AES-256-CBC:+SHA1:+MD5:+COMP-NULL", NULL);
+       gnutls_priority_set_direct(ssl_data_loc->session, "NONE:+VERS-TLS1.0:+VERS-TLS1.1:+VERS-TLS1.2:+VERS-TLS1.3:+ANON-DH:+ECDHE-RSA:+RSA:+AES-128-CBC:+AES-256-CBC:+AES-256-GCM:+SHA1:+SHA256:+SHA384:+MD5:+COMP-NULL", NULL);
        gnutls_credentials_set(ssl_data_loc->session, GNUTLS_CRD_CERTIFICATE, ssl_data_loc->certificate);
        gnutls_session_set_ptr(ssl_data_loc->session, ssl_data_loc);
```

nikias, 2019-11-25:
while this misses the version check...

hadess, 2019-11-25:
No, that doesn't work unfortunately:
`usbmuxd[53797]: [16:46:35.598][1] preflight_worker_handle_device_add: The stored pair record for device <UUID> is invalid. Removing.`