libimobiledevice/libimobiledevice #880: debugserver bug as of Nov 25th

JoshGuyette, 2019-11-26:
CC       imobiledevice_la-imobiledevice.lo
imobiledevice.c: In function ‘__pyx_f_13imobiledevice_17DebugServerClient_send_command’:
imobiledevice.c:54391:224: error: too few arguments to function ‘debugserver_client_send_command’
 ndle_error(((struct __pyx_obj_13imobiledevice_Base *)__pyx_v_self), debugserver_client_send_command(__pyx_v_self->_c_client, __pyx_v_command->_c_command, (&__pyx_v_c_response))); if (unlikely(__pyx_t_8 == ((int)-1))) __PYX_ERR(4, 122, __pyx_L6_error)
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from imobiledevice.c:634:
../include/libimobiledevice/debugserver.h:150:21: note: declared here
 debugserver_error_t debugserver_client_send_command(debugserver_client_t client, debugserver_command_t command, char** response, size_t* response_size);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
imobiledevice.c: In function ‘__pyx_f_13imobiledevice_17DebugServerClient_receive_response’:
imobiledevice.c:55506:224: error: too few arguments to function ‘debugserver_client_receive_response’
 ndle_error(((struct __pyx_obj_13imobiledevice_Base *)__pyx_v_self), debugserver_client_receive_response(__pyx_v_self->_c_client, (&__pyx_v_c_response))); if (unlikely(__pyx_t_8 == ((int)-1))) __PYX_ERR(4, 169, __pyx_L6_error)
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from imobiledevice.c:634:
../include/libimobiledevice/debugserver.h:162:21: note: declared here
 debugserver_error_t debugserver_client_receive_response(debugserver_client_t client, char** response, size_t* response_size);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:529: imobiledevice_la-imobiledevice.lo] Error 1

JoshGuyette, 2019-11-26:
Raspberry PI build not using the debian/raspbian libplist, building from yours.

JoshGuyette, 2019-11-27:
Fix is to add NULL as the final argument in debugserver_client_receive_response()... Might as well just prototype NULL as a default for the last argument, if not specified.

nightness, 2019-11-27:
Here is a patch for the cython file...
`*** libimobiledevice.orig/cython/debugserver.pxi        2019-11-27 01:32:25.000000000 -0600
--- libimobiledevice/cython/debugserver.pxi     2019-11-27 03:09:35.333994169 -0600
***************
*** 17,26 ****
      debugserver_error_t debugserver_client_free(debugserver_client_t client)
  
      debugserver_error_t debugserver_client_send(debugserver_client_t client, const char* data, uint32_t size, uint32_t *sent)
!     debugserver_error_t debugserver_client_send_command(debugserver_client_t client, debugserver_command_t command, char** response)
      debugserver_error_t debugserver_client_receive(debugserver_client_t client, char *data, uint32_t size, uint32_t *received)
      debugserver_error_t debugserver_client_receive_with_timeout(debugserver_client_t client, char *data, uint32_t size, uint32_t *received, unsigned int timeout)
!     debugserver_error_t debugserver_client_receive_response(debugserver_client_t client, char** response)
      debugserver_error_t debugserver_client_set_argv(debugserver_client_t client, int argc, char* argv[], char** response)
      debugserver_error_t debugserver_client_set_environment_hex_encoded(debugserver_client_t client, const char* env, char** response)
  
--- 17,26 ----
      debugserver_error_t debugserver_client_free(debugserver_client_t client)
  
      debugserver_error_t debugserver_client_send(debugserver_client_t client, const char* data, uint32_t size, uint32_t *sent)
!     debugserver_error_t debugserver_client_send_command(debugserver_client_t client, debugserver_command_t command, char** response, size_t* response_size)
      debugserver_error_t debugserver_client_receive(debugserver_client_t client, char *data, uint32_t size, uint32_t *received)
      debugserver_error_t debugserver_client_receive_with_timeout(debugserver_client_t client, char *data, uint32_t size, uint32_t *received, unsigned int timeout)
!     debugserver_error_t debugserver_client_receive_response(debugserver_client_t client, char** response, size_t* response_size)
      debugserver_error_t debugserver_client_set_argv(debugserver_client_t client, int argc, char* argv[], char** response)
      debugserver_error_t debugserver_client_set_environment_hex_encoded(debugserver_client_t client, const char* env, char** response)
  
***************
*** 119,125 ****
              bytes result
  
          try:
!             self.handle_error(debugserver_client_send_command(self._c_client, command._c_command, &c_response))
              if c_response:
                  result = c_response
                  return result
--- 119,125 ----
              bytes result
  
          try:
!             self.handle_error(debugserver_client_send_command(self._c_client, command._c_command, &c_response, NULL))
              if c_response:
                  result = c_response
                  return result
***************
*** 166,172 ****
              bytes result
  
          try:
!             self.handle_error(debugserver_client_receive_response(self._c_client, &c_response))
              if c_response:
                  result = c_response
                  return result
--- 166,172 ----
              bytes result
  
          try:
!             self.handle_error(debugserver_client_receive_response(self._c_client, &c_response, NULL))
              if c_response:
                  result = c_response
                  return result`

nikias, 2019-11-29:
Fixed with commit 9f79242a441ce37c28db2b84d49621d26418dc53.

closed 2019-11-29 by nikias