libimobiledevice/libimobiledevice #881: idevicebackup2 stuck at sending 'UDID'

shawnholtz, 2019-11-28:
Hello everyone. I need some help. I am trying to backup an iphone using idevicebackup2 but it stops and is stuck. I think I need a patch or something for idevicebackup2 to work with ios 13. Can someone please explain and guide me how to add the necessary patch to ubuntu 18.04 in order to successfully backup and restore my iphone in terminal?
What I get in terminal:
"idevicebackup2 --source UDID restore --system --settings --reboot /home/shawn/Documents/iphone/
Backup directory is "/home/shawn/Documents/iphone/"
Started "com.apple.mobilebackup2" service on port 49327.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting Restore...
Restoring system files: Yes
Rebooting after restore: Yes
Don't copy backup: Yes
Preserve settings of device: No
Remove items that are not restored: No
Backup password: No
Sending 'UDID/Status.plist' (189 Bytes)
Sending 'UDID/Manifest.plist' (79.7 KB)

Thanks in advance
Shawn

nikias, 2019-11-29:
Can you run idevicesyslog on another terminal? It might point out if something goes wrong. You can use grep to filter, like this:
```
idevicesyslog -u UDID |grep -i backup
```

cronzevid, 2019-12-03:
Same happens to me. Messages from idevicesyslog, while the app is stuck:
```
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (explicit): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Finished scanning domain <private> - size:149852, files:16, dirs:21, clones:0, hardlinks:0
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (explicit): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Finished scanning domain <private> - size:972048, files:16, dirs:18, clones:0, hardlinks:0
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (attribute): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (explicit): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (attribute): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (explicit): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (attribute): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Finished scanning domain <private> - size:713707, files:24, dirs:27, clones:0, hardlinks:0
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (explicit): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Finished scanning domain <private> - size:2113108, files:99, dirs:36, clones:0, hardlinks:0
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Not backed up (explicit): <private>
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Finished scanning domain <private> - size:108222, files:6, dirs:10, clones:0, hardlinks:0
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Finished scanning All domains - total size:140241165, total files:1741, total dirs:785, total clones:31, total hardlinks:0
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Scanned in 3.925 s
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Starting to prepare for preflight
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: uploadSize:3846144, uploadFileCount:60
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Free disk space on drive: 293639618560
Dec  3 22:34:14 iPhone BackupAgent2[302] <Notice>: Prepared for preflight in 0.001s
Dec  3 22:34:16 iPhone BackupAgent2[302] <Notice>: ManifestDB: closing <private>
Dec  3 22:34:16 iPhone BackupAgent2[302] <Notice>: ManifestDB: opening <private>
Dec  3 22:34:16 iPhone BackupAgent2[302] <Notice>: Uploaded 60 files of 3.85 MB in 1.029 s at 3.74 MB/s
Dec  3 22:34:17 iPhone BackupAgent2[302] <Notice>: Moved 7656 files in 1.208 s
Dec  3 22:34:17 iPhone BackupAgent2[302] <Notice>: Removed 0 items in 0.065 s
Dec  3 22:34:17 iPhone BackupAgent2[302] <Notice>: ManifestDB: closing <private>
Dec  3 22:34:17 iPhone BackupAgent2[302] <Notice>: ManifestDB: removing <private>
Dec  3 22:34:17 iPhone BackupAgent2[302] <Notice>: Backup status: <private>
Dec  3 22:34:17 iPhone BackupAgent2[302] <Notice>: Verifying backup contents
Dec  3 22:34:26 iPhone BackupAgent2(iTunesCloud)[302] <Notice>: Battery level changed. batteryLevel=0.28
Dec  3 22:36:06 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.webbookmarksd.xpc.backupActivity:EB5EB2:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:36:06 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.routined.persistence.mirroring.backup:B338B5:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:36:06 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.avatar.activity.UserRequestedBackup:AE1B76:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:36:17 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.webbookmarksd.xpc.backupActivity:EB5EB2:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:36:17 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.routined.persistence.mirroring.backup:B338B5:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:36:17 iPhone passd(CoreServicesInternal)[149] <Notice>: kExcludedFromBackupXattrName set on path: <private>
Dec  3 22:36:19 iPhone passd(CoreServicesInternal)[149] <Notice>: kExcludedFromBackupXattrName set on path: <private>
Dec  3 22:36:23 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.webbookmarksd.xpc.backupActivity:EB5EB2:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:36:23 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.routined.persistence.mirroring.backup:B338B5:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:37:10 iPhone BackupAgent2(iTunesCloud)[302] <Notice>: Battery level changed. batteryLevel=0.29
Dec  3 22:37:31 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.webbookmarksd.xpc.backupActivity:EB5EB2:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
[... cut ...]
Dec  3 22:37:42 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.webbookmarksd.xpc.backupActivity:EB5EB2:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
Dec  3 22:37:42 iPhone dasd(DuetActivitySchedulerDaemon)[136] <Notice>: com.apple.routined.persistence.mirroring.backup:B338B5:[{name: UserRequestedBackup, policyWeight: 0.100, response: {Decision: Must Not Proceed, Score: 0.00, Rationale: [{eligibleForRunning == 0}]}}
```

mexmer, 2019-12-05:
doesn't seem stuck to me

Dec  3 22:34:17 iPhone BackupAgent2[302] <Notice>: Verifying backup contents

verification of backup will take lot of time, depending on how much content has been backed up, because it will calculate hash of every file in backup, and will compare it to manifest.

for example, if your restore (until this phase) took let's say 10 minutes, you can expect as much time to be taken in this phase.

jrogers512, 2025-03-29:
How about 6 hours?  Seems excessive, especially since itunes can do it much quicker.