libimobiledevice/libimobiledevice #882: error: too few arguments to function call debugserver_client_send_command

micahstubbs, 2019-11-29:
I encounter this error running when running `make` on MacOS 10.14.6:

```zsh
➜  libimobiledevice git:(master) sudo ./configure
Configuration for libimobiledevice 1.2.1:
-------------------------------------------

  Install prefix: .........: /usr/local
  Debug code ..............: no
  Python bindings .........: yes
  SSL support backend .....: OpenSSL

  Now type 'make' to build libimobiledevice 1.2.1,
  and then 'make install' for installation.
➜  libimobiledevice git:(master) make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in common
make[2]: Nothing to be done for `all'.
Making all in src
make[2]: Nothing to be done for `all'.
Making all in include
make[2]: Nothing to be done for `all'.
Making all in cython
/Users/m/anaconda/bin/cython -I/usr/local/include/plist/cython -I../src -o imobiledevice.c imobiledevice.pyx
/Library/Developer/CommandLineTools/usr/bin/make  all-am
  CC       imobiledevice_la-imobiledevice.lo
imobiledevice.c:40869:331: error: too few arguments to function call, expected 4, have 3
  ...debugserver_client_send_command(__pyx_v_self->_c_client, __pyx_v_command->_c_command, (&__pyx_v_c_response))...
     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                            ^
../include/libimobiledevice/debugserver.h:150:1: note: 'debugserver_client_send_command' declared here
debugserver_error_t debugserver_client_send_command(debugserver_client_t client, debugserver_com...
^
imobiledevice.c:41797:306: error: too few arguments to function call, expected 3, have 2
  ...debugserver_client_receive_response(__pyx_v_self->_c_client, (&__pyx_v_c_response))); if (unlike...
     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                               ^
../include/libimobiledevice/debugserver.h:162:1: note: 'debugserver_client_receive_response' declared
      here
debugserver_error_t debugserver_client_receive_response(debugserver_client_t client, char** resp...
^
2 errors generated.
make[3]: *** [imobiledevice_la-imobiledevice.lo] Error 1
make[2]: *** [all] Error 2
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

micahstubbs, 2019-11-29:
the error seems to originate with this line https://github.com/libimobiledevice/libimobiledevice/blob/master/cython/debugserver.pxi#L20

and this line https://github.com/libimobiledevice/libimobiledevice/blob/master/cython/debugserver.pxi#L23

in `cython/debugserver.pxi`

nikias, 2019-11-29:
Fixed with commit 9f79242a441ce37c28db2b84d49621d26418dc53.

closed 2019-11-29 by nikias