libimobiledevice/libimobiledevice #883: Unable to mount developer image: Invalid value for MountPath

rmitton, 2019-12-03:
Hi,
I'm trying to mount a developer image on a iPhone SE (iOS 13.1.3, not jailbroken). However I'm getting an error back from Apple's image mounter saying the MountPath is invalid. The device has been used for debugging through Xcode in the past and everything worked fine there. I made sure to try hard-rebooting the phone but it didn't help.

I'm not too sure what's going on but it seems it doesn't like the "/Developer" mount point perhaps? Yet I've compared the commands being sent to the source code in the [SDMMobileDevice library](https://github.com/samdmarshall/SDMMobileDevice), which should I suppose be considered canon as it was reverse engineered from Apple's own code, and I can't see anything different between the two implementations.

Here's the command I'm using:

    ideviceimagemounter.exe -d s:\Downloads\DeveloperDiskImage.dmg

And here's a complete debug transcript (edited for clarity and formatting):

<?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        <key>Label</key>
        <string>ideviceimagemounter</string>
        <key>Request</key>
        <string>QueryType</string>
    </dict>
    </plist>
    -----------------------------------
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
    </dict>
    </plist>
    -----------------------------------
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        <key>Label</key>
        <string>ideviceimagemounter</string>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
    </dict>
    </plist>
    -----------------------------------
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
        <key>Value</key>
        <string>13.1.3</string>
    </dict>
    </plist>
    -----------------------------------
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        <key>Label</key>
        <string>ideviceimagemounter</string>
        <key>Request</key>
        <string>StartSession</string>
        <key>HostID</key>
        <string>30773703-1009682752452113044</string>
        <key>SystemBUID</key>
[... 150 lines cut ...]
        <string>Developer</string>
    </dict>
    </plist>
    -----------------------------------
    <?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    <plist version="1.0">
    <dict>
        <key>DetailedError</key>
        <string>Error Domain=com.apple.MobileStorage.ErrorDomain Code=-2 
        "Failed to mount /private/var/run/mobile_image_mounter/75F5581919E3F3CFBF36B599771FC383EA99ED820A011EFC4799EF4989D5A674578A85AEE06220E13E36D7DACDAF8851C5B98764E8C5467E11A1B00269EC00F2/0EB074E3B1356D7945C16F564F4C0D9A0B44F162F21D239B7477657E5BB7324803538DBC2C7C45978A5B9737516D81ECC69443CE2776F1C32701DFB58A2D60AE/xgJQmf.dmg." 
        UserInfo={
            NSLocalizedDescription=Failed to mount /private/var/run/mobile_image_mounter/75F5581919E3F3CFBF36B599771FC383EA99ED820A011EFC4799EF4989D5A674578A85AEE06220E13E36D7DACDAF8851C5B98764E8C5467E11A1B00269EC00F2/0EB074E3B1356D7945C16F564F4C0D9A0B44F162F21D239B7477657E5BB7324803538DBC2C7C45978A5B9737516D81ECC69443CE2776F1C32701DFB58A2D60AE/xgJQmf.dmg.,
            NSUnderlyingError=0x104d12ab0 {
                Error Domain=com.apple.MobileStorage.ErrorDomain Code=-2 
                "Invalid value for MountPath: Error Domain=com.apple.MobileStorage.ErrorDomain Code=-2 "Failed to mount /private/var/run/mobile_image_mounter/75F5581919E3F3CFBF36B599771FC383EA99ED820A011EFC4799EF4989D5A674578A85AEE06220E13E36D7DACDAF8851C5B98764E8C5467E11A1B00269EC00F2/0EB074E3B1356D7945C16F564F4C0D9A0B44F162F21D239B7477657E5BB7324803538DBC2C7C45978A5B9737516D81ECC69443CE2776F1C32701DFB58A2D60AE/xgJQmf.dmg at /Developer."
                UserInfo={NSLocalizedDescription=Failed to mount /private/var/run/mobile_image_mounter/75F5581919E3F3CFBF36B599771FC383EA99ED820A011EFC4799EF4989D5A674578A85AEE06220E13E36D7DACDAF8851C5B98764E8C5467E11A1B00269EC00F2/0EB074E3B1356D7945C16F564F4C0D9A0B44F162F21D239B7477657E5BB7324803538DBC2C7C45978A5B9737516D81ECC69443CE2776F1C32701DFB58A2D60AE/xgJQmf.dmg at /Developer.}" UserInfo={NSLocalizedDescription=Invalid value for MountPath: Error Domain=com.apple.MobileStorage.ErrorDomain Code=-2 "Failed to mount /private/var/run/mobile_image_mounter/75F5581919E3F3CFBF36B599771FC383EA99ED820A011EFC4799EF4989D5A674578A85AEE06220E13E36D7DACDAF8851C5B98764E8C5467E11A1B00269EC00F2/0EB074E3B1356D7945C16F564F4C0D9A0B44F162F21D239B7477657E5BB7324803538DBC2C7C45978A5B9737516D81ECC69443CE2776F1C32701DFB58A2D60AE/xgJQmf.dmg at /Developer." UserInfo={NSLocalizedDescription=Failed to mount /private/var/run/mobile_image_mounter/75F5581919E3F3CFBF36B599771FC383EA99ED820A011EFC4799EF4989D5A674578A85AEE06220E13E36D7DACDAF8851C5B98764E8C5467E11A1B00269EC00F2/0EB074E3B1356D7945C16F564F4C0D9A0B44F162F21D239B7477657E5BB7324803538DBC2C7C45978A5B9737516D81ECC69443CE2776F1C32701DFB58A2D60AE/xgJQmf.dmg at /Developer.
                }}}}
        </string>
        <key>Error</key>
        <string>ImageMountFailed</string>
    </dict>
    </plist>
    Error: ImageMountFailed
    -----------------------------------

mexmer, 2019-12-03:
it's staging.image not staging.dimage
no idea, where you got that typo.

rmitton, 2019-12-03:
Hi,
I tried changing the "dimage" to "image" in `ideviceimagemounter.c` but it gives the same error.
(also, it's not _my_ typo, it's like that in the repository) :)

rmitton, 2019-12-05:
Further investigation has revealed the cause to be a mismatch between the version of iOS running on the phone and the version of the developer disk image used. Tracking down the correct developer disk image has fixed the problem.

It's not clear why that would cause an invalid mount path error, nor if there's anything that could be done inside imobiledevice to detect it and warn the user. Presumably Xcode doesn't have this problem because Xcode dynamically picks the correct disk image to use based on the version the phone reports back.

Given that it's not a bug in imobiledevice, I'll close the issue. I still find it all very odd though.

closed 2019-12-05 by rmitton