libimobiledevice/libimobiledevice #884: Error in make command

cristiansori2, 2019-12-03:
All libraries installed,libusbmuxd installed on version 2.0.1,using Ubuntu 18.04.3:

> idevice.c: In function ‘usbmux_event_cb’:
idevice.c:194:19: error: ‘usbmuxd_device_info_t {aka const struct <anonymous>}’ has no member named ‘conn_type’
  if (event->device.conn_type == CONNECTION_TYPE_USB) {
                   ^
idevice.c:194:33: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function)
  if (event->device.conn_type == CONNECTION_TYPE_USB) {
                                 ^~~~~~~~~~~~~~~~~~~
idevice.c:194:33: note: each undeclared identifier is reported only once for each function it appears in
idevice.c:196:26: error: ‘usbmuxd_device_info_t {aka const struct <anonymous>}’ has no member named ‘conn_type’
  } else if (event->device.conn_type == CONNECTION_TYPE_NETWORK) {
                          ^
idevice.c:196:40: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function)
  } else if (event->device.conn_type == CONNECTION_TYPE_NETWORK) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
In file included from idevice.c:45:0:
idevice.c:199:57: error: ‘usbmuxd_device_info_t {aka const struct <anonymous>}’ has no member named ‘conn_type’
   debug_info("Unknown connection type %d", event->device.conn_type);
                                                         ^
../common/debug.h:29:72: note: in definition of macro ‘debug_info’
 fine debug_info(...) debug_info_real (__func__, __FILE__, __LINE__, __VA_ARGS__)
                                                                     ^~~~~~~~~~~
idevice.c: In function ‘idevice_get_device_list_extended’:
idevice.c:249:18: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                  ^
idevice.c:249:32: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function)
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                                ^~~~~~~~~~~~~~~~~~~
idevice.c:252:25: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
   } else if (dev_list[i].conn_type == CONNECTION_TYPE_NETWORK) {
                         ^
idevice.c:252:39: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function)
   } else if (dev_list[i].conn_type == CONNECTION_TYPE_NETWORK) {
                                       ^~~~~~~~~~~~~~~~~~~~~~~
idevice.c:254:32: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
    size_t addrlen = dev_list[i].conn_data[0];
                                ^
idevice.c:256:52: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
    memcpy(newlist[newcount]->conn_data, dev_list[i].conn_data, addrlen);
                                                    ^
idevice.c: In function ‘idevice_get_device_list’:
idevice.c:302:18: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                  ^
idevice.c:302:32: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function)
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                                ^~~~~~~~~~~~~~~~~~~
idevice.c: In function ‘idevice_from_mux_device’:
idevice.c:348:16: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
  switch (muxdev->conn_type) {
                ^~
idevice.c:349:7: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function)
  case CONNECTION_TYPE_USB:
       ^~~~~~~~~~~~~~~~~~~
idevice.c:353:7: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function)
  case CONNECTION_TYPE_NETWORK:
       ^~~~~~~~~~~~~~~~~~~~~~~
idevice.c:355:33: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
   size_t len = ((uint8_t*)muxdev->conn_data)[0];
                                 ^~
idevice.c:357:35: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
   memcpy(device->conn_data, muxdev->conn_data, len);
                                   ^~
idevice.c:348:2: warning: switch missing default case [-Wswitch-default]
  switch (muxdev->conn_type) {
  ^~~~~~
idevice.c: In function ‘idevice_new_with_options’:
idevice.c:372:21: error: ‘DEVICE_LOOKUP_USBMUX’ undeclared (first use in this function)
   usbmux_options |= DEVICE_LOOKUP_USBMUX;
                     ^~~~~~~~~~~~~~~~~~~~
idevice.c:375:21: error: ‘DEVICE_LOOKUP_NETWORK’ undeclared (first use in this function)
   usbmux_options |= DEVICE_LOOKUP_NETWORK;
                     ^~~~~~~~~~~~~~~~~~~~~
idevice.c:378:21: error: ‘DEVICE_LOOKUP_PREFER_NETWORK’ undeclared (first use in this function)
   usbmux_options |= DEVICE_LOOKUP_PREFER_NETWORK;
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
idevice.c:380:12: warning: implicit declaration of function ‘usbmuxd_get_device’ [-Wimplicit-function-declaration]
  int res = usbmuxd_get_device(udid, &muxdev, usbmux_options);
            ^~~~~~~~~~~~~~~~~~
Makefile:549: recipe for target 'idevice.lo' failed
make[2]: *** [idevice.lo] Error 1
make[2]: Leaving directory '/home/cristian/libimobiledevice/src'
Makefile:454: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/cristian/libimobiledevice'
Makefile:384: recipe for target 'all' failed
make: *** [all] Error 2

closed 2019-12-03 by cristiansori2

wecreate1, 2019-12-15:
Can you explain how you fixed this issue? I'm having a similar one.

anirath, 2020-01-05:
Same here. I can't seem to figure it out either. I went through and verified that I've installed all of the necessary dependencies, and especially the usbmuxd related ones. I don't get it...