libimobiledevice/libimobiledevice #891: OpenSSL 1.1.1d  compilation issues

Dantee296, 2019-12-19:
how to fix these issues

Undefined symbols for architecture x86_64:
  "_X509_set1_notAfter", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_X509_set1_notBefore", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

closed 2019-12-19 by Dantee296

Parabak, 2020-03-16:
Hi @Dantee296 , I have the same issue. Did you find the reason?
Thanks.

guyingzhao, 2020-03-30:
export openssl_LIBS="-L/usr/local/Cellar/openssl/1.0.2s/lib/ -lssl -lcrypto -lz"
export openssl_CFLAGS=-I/usr/local/Cellar/openssl/1.0.2s/include/

I solved this by export variable above