libimobiledevice/libimobiledevice #893: idevicedebug sometimes fails to start app with '�' output

jhihn, 2019-12-20:
Only works ~33 % of the time:

    $ idevicedebug -d run com.yadda.yadda
    �

5 successes
9 failures

But the app does not launch. When it works, it works without printing anything. It does not matter if the screen id off or on. Also does not matter if the app was already launched.

jpstotz, 2020-02-09:
I can reproduce the problem especially the nondeterministic behavior. 

Based on my experience the timeout of 1000 ms compiled into idevicedebug is way too low in certain situations. Additionally idevicedebug does not properly check if it runs into a timeout or not. Instead it just continues which sometimes work but sometimes not. 

I have prepared pull request #906 that applies proper error handling to the startup phase of idevicedebug. Unfortunately I don't get any response from the maintainers of libimobiledevice.

jhihn, 2020-02-09:
Truly excellent. 
Last time I found an issue they were plenty responsive. Maybe not many people use this so its not in their radar...?

jpstotz, 2020-02-09:
The only active maintainer at libimobiledevice seems to be @nikias and looking at all the available pull requests (and especially the closed one) I get that feeling that he does not care abut PRs. 

The number of applied PRs in the last years is very very small. Usually a PR is closed by the author because of various reasons or it remains open usually without any comment.
May be the code quality is not sufficient from Nikias's perspective but in this is the case it would be nice of him telling this to the authors of the PRs so that they can improve it.

jhihn, 2020-02-09:
Previously I got some help from @seplu. Forgive me if this is bad form mentioning you here seplu. @jpstotz your patch seems simple and clean. I also agree with your suggestion extending the timeout and making it a command line parameter.