libimobiledevice/libimobiledevice #898: Error during autogen on MacOS 10.15.2 (19C57)

abraunstein, 2020-01-03:
New to GitHub, so be gentle. :)

Pretty simple situation.  I downloaded the zip of this project and as per the instructions ran autogen.sh.  The output (included following) seems to indicate that things go as expected until the end - at which point it dies because openssl can't be found.  Seeing as openssl is a pretty important component, the install ends up being unusable.

I know that Apple has been monkeying around with ssl-related components of late - which means that my first suspicion is that something changed in Catalina that causes autogen to fail.

Thoughts?  Suggestions?  Commiseration?

Thanks!
Aaron

glibtoolize: putting auxiliary files in '.'.
glibtoolize: linking file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: linking file 'm4/libtool.m4'
glibtoolize: linking file 'm4/ltoptions.m4'
glibtoolize: linking file 'm4/ltsugar.m4'
glibtoolize: linking file 'm4/ltversion.m4'
glibtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:31: installing './compile'
configure.ac:34: installing './config.guess'
configure.ac:34: installing './config.sub'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
common/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-apple-darwin19.2.0
checking host system type... x86_64-apple-darwin19.2.0
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
checking how to convert x86_64-apple-darwin19.2.0 file names to x86_64-apple-darwin19.2.0 format... func_convert_file_noop
checking how to convert x86_64-apple-darwin19.2.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking for -force_load linker flag... yes
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... yes
checking for gcc option to produce PIC... -fno-common -DPIC
checking if gcc PIC flag -fno-common -DPIC works... yes
checking if gcc static flag -static works... no
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/Library/Developer/CommandLineTools/usr/bin/ld) supports shared libraries... yes
checking dynamic linker characteristics... darwin19.2.0 dyld
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking whether the g++ linker (/Library/Developer/CommandLineTools/usr/bin/ld) supports shared libraries... yes
checking for g++ option to produce PIC... -fno-common -DPIC
checking if g++ PIC flag -fno-common -DPIC works... yes
checking if g++ static flag -static works... no
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/Library/Developer/CommandLineTools/usr/bin/ld) supports shared libraries... yes
checking dynamic linker characteristics... darwin19.2.0 dyld
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/local/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd >= 1.1.0... yes
checking for libplist >= 1.11... yes
checking for libplist++ >= 1.11... yes
checking for ANSI C header files... (cached) yes
checking for stdint.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking gcrypt.h usability... no
checking gcrypt.h presence... no
checking for gcrypt.h... no
checking for an ANSI C-conforming const... yes
checking for size_t... yes
checking for ssize_t... yes
checking for uint16_t... yes
checking for uint32_t... yes
checking for uint8_t... yes
checking for asprintf... yes
checking for strcasecmp... yes
checking for strdup... yes
checking for strerror... yes
checking for strndup... yes
checking for stpcpy... yes
checking for vasprintf... yes
checking endian.h usability... no
checking endian.h presence... no
checking for endian.h... no
checking whether byte ordering is bigendian... no
checking for platform-specific build settings... darwin19.2.0
checking for struct dirent.d_type... yes
checking for a Python interpreter with version >= 2.3... python
checking for python... /usr/bin/python
checking for python version... 2.7
checking for python platform... darwin
checking for python script directory... ${prefix}/lib/python2.7/site-packages
checking for python extension module directory... ${exec_prefix}/lib/python2.7/site-packages
checking for cython... no
configure: WARNING: cannot find 'cython' program. You should look at http://www.cython.org or install your distribution specific cython package.
checking for python2.7... (cached) /usr/bin/python
checking for a version of Python >= '2.1.0'... yes
checking for the distutils Python package... yes
checking for Python include path... -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7
checking for Python library path... -L/System/Library/Frameworks/Python.framework/Versions/2.7/lib -lpython2.7
checking for Python site-packages path... /Library/Python/2.7/site-packages
checking python extra libraries... -ldl  -framework CoreFoundation 
checking python extra linking flags... -u _PyMac_Error /System/Library/Frameworks/Python.framework/Versions/2.7/Python
checking consistency of all components of python development environment... yes
checking for openssl >= 0.9.8... no
configure: error: OpenSSL support explicitly requested but OpenSSL could not be found

qmfrederik, 2020-01-14:
Did you install OpenSSL on your Mac? You can use Homebrew to do so, via `brew install openssl`.

schlachtzeuger, 2020-02-10:
I'm facing the same issue and have OpenSSL 1.1.1d installed via Brew.

schlachtzeuger, 2020-02-10:
Found the answer here: https://github.com/Homebrew/homebrew-core/issues/47515#issuecomment-562196419