libimobiledevice/libimobiledevice #899: Never Finishes Sending Status.plist' and Manifest.plist'

kendoori, 2020-01-05:
Trying to do a full backup of my iPad Mini 4 via Linux Mint 19.3. I used this command: idevicebackup2 backup /media/kenny/X230-MINT-BACKUP/iPad-Backup 

It moves through the file list fairly quickly and I can see the file system and assets built on my target side, but it doesn't seem to finish. I let it run the first time for 12 hours, and the behavior is consistent.

Not sure if it matters, but the target device is an SD card mounted on my laptop.

[image]

cerritus55, 2020-01-05:
I got the same issue, as kendoori.  Trying to to a full backub of my Iphone 7 with  Ubuntu 18.04.3 LTS x86_64
It moves through the file list fairly quickly but then it never finish. I let it run for 6 hours.

dylanatgroundwater, 2020-03-15:
Same problem; mint 19.3, iphone se, ios 13.3. Additionally, if I try to kill the process I get an 'exiting...' response though it never actually 'exits'. Will try on debian based distribution..

jrogers512, 2025-03-29:
I have the same symptoms on 24.04.