libimobiledevice/libimobiledevice #900: Possible bug surfacing in rvi_capture

andrewmusselman, 2020-01-06:
https://github.com/gh2o/rvi_capture/issues/4

I'm encountering this bug where a packet capture utility says it has "Not Enough Data" which seems to occur in this library. Any ideas?

On Ubuntu with a device running iOS 13.3.

```Traceback (most recent call last):
  File "./rvi_capture.py", line 364, in <module>
    main()
  File "./rvi_capture.py", line 354, in main
    packet_dumper.run(packet_callback)
  File "./rvi_capture.py", line 295, in run
    for pkt in self.pkt_iter:
  File "./rvi_capture.py", line 183, in __iter__
    [chunk_len] = UB32.unpack(read_fully(4))
  File "./rvi_capture.py", line 178, in read_fully
    b += conn.recv(n - l)
  File "./rvi_capture.py", line 123, in recv
    self.handle, out, num_bytes, ctypes.byref(out_bytes), ctypes.c_uint(500)))
  File "./rvi_capture.py", line 59, in check
    raise cls(err)
__main__.IDeviceError: Error in libimobiledevice: Not Enough Data
```

oi23mkoshiba, 2020-03-31:
I'm Japanese, so I apologize for my poor English.

I got this bug, too.
But my packet capture utility says it has "Missing Error".

On Fedora 31 with device running iOS 12.4.5
```
Traceback (most recent call last):
  File "./debugtest.py", line 371, in <module>
    main()
  File "./debugtest.py", line 361, in main
    packet_dumper.run(packet_callback)
  File "./debugtest.py", line 295, in run
    for pkt in self.pkt_iter:
  File "./debugtest.py", line 183, in __iter__
    [chunk_len] = UB32.unpack(read_fully(4))
  File "./debugtest.py", line 178, in read_fully
    b += conn.recv(n - l)
  File "./debugtest.py", line 123, in recv
    self.handle, out, num_bytes, ctypes.byref(out_bytes), ctypes.c_uint(500)))
  File "./debugtest.py", line 59, in check
    raise cls(err)
__main__.IDeviceError: Error in libimobiledevice: Missing Error

```

N8TFly, 2020-09-22:
@oi23mkoshiba did you manage to get it working?

oi23mkoshiba, 2020-09-22:
> @oi23mkoshiba did you manage to get it working? I have the same error

@N8TFly 
Sorry, I use translate.google.com.
I can now use rvi_capture.
The cause was "usbmuxd" running inside the program.
The new version of "usbmuxd" had a bug that didn't load the iPhone well.
So, when I downgraded "usbmuxd", rvi_capture worked fine.
In the Fedora, I ran the command: `sudo dnf downgrade usbmuxd`

mexmer, 2020-09-23:
if rvi doesn't work with latest usbmux, it means rvi needs to be upgraded

gh2o, 2021-11-15:
Issue was in rvi_capture and was solved, please close this bug.