libimobiledevice/libimobiledevice #901: Error in make on Ubuntu 18.04

aquaductape, 2020-01-08:
Using Ubuntu 18.04. Not sure what to do since I have all the software listed in the README, except with libtool, I installed libtool-bin.

usbmuxd 
make
autoheader
automake
autoconf
libtool-bin
pkg-config
gcc or clang

```
make  all-recursive
make[1]: Entering directory '/root/libimobiledevice'
Making all in common
make[2]: Entering directory '/root/libimobiledevice/common'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/libimobiledevice/common'
Making all in src
make[2]: Entering directory '/root/libimobiledevice/src'
  CC       idevice.lo
idevice.c: In function ‘usbmux_event_cb’:
idevice.c:198:19: error: ‘usbmuxd_device_info_t {aka const struct <anonymous>}’ has no member named ‘conn_type’
  if (event->device.conn_type == CONNECTION_TYPE_USB) {
                   ^
idevice.c:198:33: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
  if (event->device.conn_type == CONNECTION_TYPE_USB) {
                                 ^~~~~~~~~~~~~~~~~~~
                                 CONNECTION_USBMUXD
idevice.c:198:33: note: each undeclared identifier is reported only once for each function it appears in
idevice.c:200:26: error: ‘usbmuxd_device_info_t {aka const struct <anonymous>}’ has no member named ‘conn_type’
  } else if (event->device.conn_type == CONNECTION_TYPE_NETWORK) {
                          ^
idevice.c:200:40: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function); did you mean ‘CONNECTION_NETWORK’?
  } else if (event->device.conn_type == CONNECTION_TYPE_NETWORK) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
                                        CONNECTION_NETWORK
idevice.c: In function ‘idevice_get_device_list_extended’:
idevice.c:253:18: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                  ^
idevice.c:253:32: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                                ^~~~~~~~~~~~~~~~~~~
                                CONNECTION_USBMUXD
idevice.c:256:25: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
   } else if (dev_list[i].conn_type == CONNECTION_TYPE_NETWORK) {
                         ^
idevice.c:256:39: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function); did you mean ‘CONNECTION_NETWORK’?
   } else if (dev_list[i].conn_type == CONNECTION_TYPE_NETWORK) {
                                       ^~~~~~~~~~~~~~~~~~~~~~~
                                       CONNECTION_NETWORK
idevice.c:258:32: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
    size_t addrlen = dev_list[i].conn_data[0];
                                ^
idevice.c:260:52: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
    memcpy(newlist[newcount]->conn_data, dev_list[i].conn_data, addrlen);
                                                    ^
idevice.c: In function ‘idevice_get_device_list’:
idevice.c:306:18: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                  ^
idevice.c:306:32: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
   if (dev_list[i].conn_type == CONNECTION_TYPE_USB) {
                                ^~~~~~~~~~~~~~~~~~~
                                CONNECTION_USBMUXD
idevice.c: In function ‘idevice_from_mux_device’:
idevice.c:352:16: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_type’
  switch (muxdev->conn_type) {
                ^~
idevice.c:353:7: error: ‘CONNECTION_TYPE_USB’ undeclared (first use in this function); did you mean ‘CONNECTION_USBMUXD’?
  case CONNECTION_TYPE_USB:
       ^~~~~~~~~~~~~~~~~~~
       CONNECTION_USBMUXD
idevice.c:357:7: error: ‘CONNECTION_TYPE_NETWORK’ undeclared (first use in this function); did you mean ‘CONNECTION_NETWORK’?
  case CONNECTION_TYPE_NETWORK:
       ^~~~~~~~~~~~~~~~~~~~~~~
       CONNECTION_NETWORK
idevice.c:359:33: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
   size_t len = ((uint8_t*)muxdev->conn_data)[0];
                                 ^~
idevice.c:361:35: error: ‘usbmuxd_device_info_t {aka struct <anonymous>}’ has no member named ‘conn_data’
   memcpy(device->conn_data, muxdev->conn_data, len);
                                   ^~
idevice.c:352:2: warning: switch missing default case [-Wswitch-default]
  switch (muxdev->conn_type) {
  ^~~~~~
idevice.c: In function ‘idevice_new_with_options’:
idevice.c:376:21: error: ‘DEVICE_LOOKUP_USBMUX’ undeclared (first use in this function); did you mean ‘IDEVICE_LOOKUP_USBMUX’?
   usbmux_options |= DEVICE_LOOKUP_USBMUX;
                     ^~~~~~~~~~~~~~~~~~~~
                     IDEVICE_LOOKUP_USBMUX
idevice.c:379:21: error: ‘DEVICE_LOOKUP_NETWORK’ undeclared (first use in this function); did you mean ‘IDEVICE_LOOKUP_NETWORK’?
   usbmux_options |= DEVICE_LOOKUP_NETWORK;
                     ^~~~~~~~~~~~~~~~~~~~~
                     IDEVICE_LOOKUP_NETWORK
idevice.c:382:21: error: ‘DEVICE_LOOKUP_PREFER_NETWORK’ undeclared (first use in this function); did you mean ‘IDEVICE_LOOKUP_PREFER_NETWORK’?
   usbmux_options |= DEVICE_LOOKUP_PREFER_NETWORK;
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                     IDEVICE_LOOKUP_PREFER_NETWORK
idevice.c:384:12: warning: implicit declaration of function ‘usbmuxd_get_device’; did you mean ‘usbmuxd_get_device_list’? [-Wimplicit-function-declaration]
  int res = usbmuxd_get_device(udid, &muxdev, usbmux_options);
            ^~~~~~~~~~~~~~~~~~
            usbmuxd_get_device_list
Makefile:546: recipe for target 'idevice.lo' failed
make[2]: *** [idevice.lo] Error 1
make[2]: Leaving directory '/root/libimobiledevice/src'
Makefile:451: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/root/libimobiledevice'
Makefile:381: recipe for target 'all' failed
make: *** [all] Error 2
```

nikias, 2020-01-08:
You need a newer version of libusbmuxd from source installed.

aquaductape, 2020-01-08:
Install latest [libusbmuxd from here](https://launchpad.net/ubuntu/+source/libusbmuxd/2.0.1-2)?

aquaductape, 2020-01-08:
Solved! I didn't know that [libusbmuxd](https://github.com/libimobiledevice/libusbmuxd) and [libplist](https://github.com/libimobiledevice/libplist) were repos that belonged to libimobiledevice so I installed them. I had to install libplist to update libplist first because libusbmuxd required libplist 2.1.0.`

closed 2020-01-08 by aquaductape