libimobiledevice/libimobiledevice #902: error on idevice_id -l

aquaductape, 2020-01-09:
I'm using Ubuntu 18.04, I was able to install ios-webkit-debug-proxy and libimobiledevice without issues. When running `idevice_id -l`  I get an error:
```
idevice_id: symbol lookup error: idevice_id: undefined symbol: idevice_device_list_extended_free
```

I'll the share the logs of the steps I took.

1. [sudo apt-get install autoconf automake libusb-dev libusb-1.0-0-dev libplist-dev libplist++-dev usbmuxd libtool libimobiledevice-dev libssl-dev](https://docs.google.com/document/d/1HaCQ1IL1p3kzVVmAZJ1dEX3MxvWiHkjWGkJEFiVv_R0/edit?usp=sharing)

2. [ios-webkit-debug-proxy](https://docs.google.com/document/d/1x6Hz2MZGRi0zyn4_7vt16ZCbO_daBzyB2oqmkVjXpLU/edit?usp=sharing)

3. [libplist](https://docs.google.com/document/d/1uxoo5X3ZMH_4h76N7djVdvtpWq3HbJo5hL1PhNTrj3E/edit?usp=sharing)

4. [libusbmuxd](https://docs.google.com/document/d/10WuUbUyu8X8AVjm42uFQOv7n1JJMLDpvnlRKT9LKlSs/edit?usp=sharing)

5. [libimobiledevice](https://docs.google.com/document/d/1WQnpK7EQuJE03dP7BqHYDYhUx6lfEdpSLf5uT7gk5vs/edit?usp=sharing)

6. [sudo apt-get install libimobiledevice-utils](https://docs.google.com/document/d/1HSl0QVZgn8yRJausp7ZBNsOSihxjNl0PO6VyXPHEt1g/edit?usp=sharing)

CodeWithOz, 2020-03-03:
@qmfrederik because the [instructions for `ios-webkit-debug-proxy`](https://github.com/google/ios-webkit-debug-proxy#usage) say so. Looks to me like one can expect the ubuntu repositories to be out of date so perhaps the `ios-webkit-debug-proxy` docs should indicate this and suggest building everything from source. Ultimately this is not a problem with this library.

mattparrilla, 2020-03-13:
@CodeWithOz perhaps not a problem with the library, but I'm in the same boat for the same reason! What did you do to resolve?