libimobiledevice/libimobiledevice #903: Idevicebackup2 fails on ios 13.2.3 and ios13.3

Overc1ocker, 2020-01-10:
Hello everyone! I just learned that you can back up iphones on linux with a simple terminal command, however when I run the command `idevicebackup2 backup /home/linux-pc/iphonebackups`, I get a "device not ready. Retrying in 2 seconds" error. I have already tried the idevicepair pair command and specifying udid. Idevicepair says my phone is connected. I have tried 2 phones so far. is there anything else I can try?

Package version: libimobiledevice 1.2.0+87+g92c5462-6 (arch linux, manjaro)

Jackfritt, 2020-01-17:
I had the same problem. Try to compile from latest git and try that version.
Maybe this helps.

git clone https://github.com/libimobiledevice/libplist.git
cd libplist/
./autogen.sh
make
sudo make install
cd ..

git clone https://github.com/libimobiledevice/libusbmuxd.git
cd libusbmuxd/
./autogen.sh
make
sudo make install
cd ..

git clone https://github.com/libimobiledevice/libimobiledevice.git
cd libimobiledevice/
./autogen.sh
make
sudo make install
cd ..

tools should be in 
/usr/local/bin/

Is the iphone still activated? You need to do that. Try the WLAN Option

What says
/usr/local/bin/ideviceinfo

ActivationState: Unactivated?
then you need to activate.

Overc1ocker, 2020-01-18:
That worked! Thanks! I just made my first ever iphone backup without itunes!

closed 2020-01-18 by Overc1ocker

yoonsikp, 2020-03-18:
Thanks, also needed to compile and start usbmuxd