libimobiledevice/libimobiledevice #905: Build problems: "undeclared identifier" and "has no member named"

liam-clink, 2020-01-19:
[make errors.txt](https://github.com/libimobiledevice/libimobiledevice/files/4083508/make.errors.txt)

I have attached the problematic output. Pretty much all of the errors are from the compiler and are along the lines of "this is undeclared, did you mean this?" and "this struct has no member named that"

nikias, 2020-01-19:
You don't have latest libusbmuxd.

liam-clink, 2020-01-19:
Which version is required? I have 1.1.0 installed

mattwalkenhorst, 2020-01-19:
@liam-clink I just had the same build issues. I was able build by cloning and building the dependencies as listed in [this issue](https://github.com/libimobiledevice/libimobiledevice/issues/903)

nikias, 2020-01-20:
@liam-clink you need >= 2.0 but just clone the latest from git: https://github.com/libimobiledevice/libusbmuxd