libimobiledevice/libimobiledevice #910: Any plans to make new release?

kloczek, 2020-02-10:
I think that it would be good to flush currently committed changes and make new release :)

James-Hudson3010, 2020-02-14:
Yes,  please. I have been successfully using the master branches for my needs for awhile now. The last official version stopping working for me awhile ago.

LaikaN57, 2020-03-11:
Agreed. Normally I would say this does not classify as a GH issue but there are a lot of projects that only run off stable tags.

Overc1ocker, 2020-03-24:
Expecting distros to just clone master every few months makes you run into this issue... https://forum.manjaro.org/t/libimobiledevice-is-out-of-date-solutions/131366/8

LaikaN57, 2020-03-24:
This is a duplicate of #837 and should be closed.

FunkyM, 2020-06-09:
Releases of all projects around libimobiledevice hosted here are imminent.
A regular release schedule is going to be setup by the team soon.

closed 2020-06-09 by FunkyM