libimobiledevice/libimobiledevice #916: missing conn_type member in usbmuxd_device_info_t error generated during ./autogen

James-Hudson3010, 2020-02-13:
`configure: error: libusbmuxd (usbmuxd.h) is not up-to-date; missing conn_type member in usbmuxd_device_info_t. Please update libusbmuxd to build libimobiledevice`

I have built and installed the master branch of https://github.com/libimobiledevice/libusbmuxd.git and it can be found via pkg-config.

Perhaps something hasn't been pushed?

James-Hudson3010, 2020-02-13:
looking at this a bit more, 61babf5 added this check.

I can go back to 297ae43 and make without error.

nikias, 2020-02-14:
It does work:
```
checking for usbmuxd_device_info_t.conn_type... yes
```
It might find system headers that it takes over the newly installed libusbmuxd headers. What system are you on?

James-Hudson3010, 2020-02-14:
macOS X 10.15.3 with the latest version of Xcode installed.

I did install https://github.com/libimobiledevice/libusbmuxd from source and pkg-config does find it correctly.

```
$ pkg-config --list-all
...
libusbmuxd                libusbmuxd - A library to communicate with the usbmux daemon
...

$ pkg-config --cflags libusbmuxd
-I/Users/james/local/include

$ pkg-config --libs libusbmuxd
-L/Users/james/local/lib -lusbmuxd

$ pkg-config --modversion libusbmuxd
2.0.2
```

zloss, 2020-02-20:
The function "ac_fn_c_try_compile" in "configure" didn't search the custom include path for "usbmuxd.h",  so you need to copy the header files of "libusbmuxd"  to /usr/local/include ,then it works.

> `configure: error: libusbmuxd (usbmuxd.h) is not up-to-date; missing conn_type member in usbmuxd_device_info_t. Please update libusbmuxd to build libimobiledevice`
>
> I have built and installed the [...]

James-Hudson3010, 2020-02-20:
That would certainly appear to work around the problem for now. Hopefully something can be changed so that the build will look in the place that pkg-config says to look.

nikias, 2020-02-20:
@James-Hudson3010 I am looking into it. Need to have it accept CFLAGS for the try compile check.

nikias, 2020-02-21:
@James-Hudson3010 can you move the header from /usr/local/include, apply this patch and try if it finds it?
```diff
diff --git a/configure.ac b/configure.ac
index 7041667..7a4ffd4 100644
--- a/configure.ac
+++ b/configure.ac
@@ -81,6 +81,11 @@ case ${host_os} in
 esac
 AM_CONDITIONAL(WIN32, test x$win32 = xtrue)

+# save CFLAGS
+old_CFLAGS="$CFLAGS"
+# ask pkg-config where libusbmuxd's headers are
+CFLAGS="-I$($PKG_CONFIG --variable=includedir libusbmuxd) $CFLAGS"
+
 AC_CHECK_MEMBER(usbmuxd_device_info_t.conn_type,, AC_MSG_ERROR([libusbmuxd (usbmuxd.h) is not up-to-date; missing conn_type member in usbmuxd_device_info_t. Please update libusbmuxd to build $PACKAGE_NAME]), [#include <usbmuxd.h>])
 AC_TRY_COMPILE([
   #include <usbmuxd.h>
@@ -94,6 +99,9 @@ AC_TRY_COMPILE([
   enum usbmux_lookup_options opts = DEVICE_LOOKUP_USBMUX | DEVICE_LOOKUP_NETWORK | DEVICE_LOOKUP_PREFER_NETWORK;
 ],, AC_MSG_ERROR([[libusbmuxd (usbmuxd.h) is not up-to-date; missing enum usbmux_lookup_options with DEVICE_LOOKUP_USBMUX, DEVICE_LOOKUP_NETWORK, and DEVICE_LOOKUP_PREFER_NETWORK - Please update libusbmuxd to build $PACKAGE_NAME]]))

+# reset CFLAGS
+CFLAGS="$old_CFLAGS"
+
 AC_CHECK_MEMBER(struct dirent.d_type, AC_DEFINE(HAVE_DIRENT_D_TYPE, 1, [define if struct dirent has member d_type]),, [#include <dirent.h>])

 # Cython Python Bindings
```

James-Hudson3010, 2020-02-21:
yes, that appears to work. ./autogen.sh worked without error and I was able to make and install the master branch after applying the patch.

nikias, 2020-02-22:
@James-Hudson3010 So what if you undo the patch and run this:
```
CFLAGS="-I/Users/james/local/include" ./autogen.sh
```
I think that would be the correct way to handle non-standard location installs.

James-Hudson3010, 2020-02-24:
That works.

It still seems to me that pkg-config, which contains that same information (-I/Users/james/local/include), should be the one who tells the build system where to find headers. That is the job of pkg-config. Why can it not be used?

tng10, 2020-02-27:
Hi, I also have the same issue here on Ubuntu 18.04

When running `./autogen.sh` I have the following
```
...
checking for usbmuxd_device_info_t.conn_type... no
configure: error: libusbmuxd (usbmuxd.h) is not up-to-date; missing conn_type member in usbmuxd_device_info_t. Please update libusbmuxd to build libimobiledevice
```

`pkg-config` output

```
$ pkg-config --list-all | grep libusbmuxd
libusbmuxd                libusbmuxd - A library to communicate with the usbmux daemon
```

```
$ pkg-config --cflags libusbmuxd
(nothing)
```

```
$ pkg-config --cflags libusbmuxd
-lusbmuxd
```

```
pkg-config --modversion libusbmuxd
1.1.0
```

nikias, 2020-02-27:
Yes you need a newer version of libusbmuxd built from git. The one displayed is the one that comes with ubuntu.