libimobiledevice/libimobiledevice #919: Is Wifi Sync only supported via iTunes provided usbmuxd?

sprhawk, 2020-02-28:
I cannot find devices via `idevice_id -n`, but I saw some issues related to wifi sync. So I guess it is only supported via Apple provided usbmuxd service?

`ideviceinfo -n -q com.apple.mobile.wireless_lockdown`
will get following:
```
BonjourFullServiceName: dd:1c:79:33:2d:1c@fe80::ec5d:02ce:de16:3ce2._apple-mobdev2._tcp.local.
EnableWifiDebugging: false
SupportsWifi: true
SupportsWifiSyncing: true
```

And I quickly browse source code in usbmuxd, it seems it doesn't have network / wifi sync related code.

mohammad-ghassemi, 2020-04-23:
Try to use https://github.com/tihmstar/usbmuxd2.

sprhawk, 2020-04-24:
> Try to use https://github.com/tihmstar/usbmuxd2.

Cool, thanks

closed 2020-04-24 by sprhawk

Tkd-Alex, 2020-12-15:
Any tutorial or documentation about usbmuxd2 and libimobiledevice ?

mohammad-ghassemi, 2020-12-15:
Not any that I know of! You can check this Twitter thread:
https://twitter.com/tihmstar/status/1171829247729647616